zephyrproject-rtos/zephyr #37376: samples/subsys/usb/dfu/sample.usb.dfu fails on teensy41/teensy40

galak, 2021-08-02:
`./scripts/twister -T samples/subsys/usb/dfu/ -p teensy41 -p teensy40`

Gets:

```

warning: FLASH_PAGE_LAYOUT (defined at drivers/flash/Kconfig:54) has direct dependencies FLASH_HAS_PAGE_LAYOUT && FLASH with value n, but is currently being y-selected by the following symbols:
 - STREAM_FLASH (defined at subsys/storage/stream/Kconfig:7), with value y, direct dependencies y (value: y)

warning: FLASH_MAP (defined at subsys/storage/flash_map/Kconfig:10) has direct dependencies FLASH_HAS_DRIVER_ENABLED with value n, but is currently being y-selected by the following symbols:
 - MCUBOOT_IMG_MANAGER (defined at subsys/dfu/Kconfig:25), with value y, direct dependencies <choice> (value: y), and select condition <choice> (value: y)

error: Aborting due to Kconfig warnings

```

galak, 2021-08-02:
@bdkrae can you please look into this.

Laczen, 2021-08-02:
@galak, these boards do not have partitions defined. The test should not run, maybe something wrong with the detection of fixed-partition, slot0 and slot1.

PR #37382 (2021-08-02): board: arm: fix failing test on teensy boards
Closes #37376
closed 2021-08-02 by galak