zephyrproject-rtos/zephyr #37390: nRF Zephyr Timing SoC implementation

davidelogi, 2021-08-03:
**Describe the bug**
I'm playing with the timing subsystem in order to measure some execution time with higher granularity than the RTC tick on nRF52840 (using PCA10056 DK).

I'm specifically using the Nordic SoC implementation (NOT the architecture default for all Cortex-M using DWT):
```
...
CONFIG_TIMING_FUNCTIONS=y
CONFIG_CORTEX_M_DWT=n
...
```
This means that the backend is using this file https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/nordic_nrf/timing.c

The TIMER instance is initialised with 16MHZ frequency and 32-bit resolution.

When getting the cycle count over a period, the developer needs to use this function:
```
uint64_t soc_timing_cycles_get(volatile timing_t *const start,
			       volatile timing_t *const end)
{
#if defined(CONFIG_SOC_SERIES_NRF51X)
#define COUNTER_SPAN BIT(16)
	if (*end >= *start) {
		return (*end - *start);
	} else {
		return COUNTER_SPAN + *end - *start;
	}
#else
	return (*end - *start);
#endif
}
```
If you try to detect average execution delta over time, you of course expect to end up in the situation I described, where:
start = large number, but coming first in time
end  = small number, but coming last in time

What I think is happening is that overflow is not on UINT64_MAX boundary, but on a smaller value, as the max value you can have using soc_timing_counter_get is:

UINT32_MAX * 4 (which is << UINT64_MAX)

This is because by nature, these are just 32-bit values (as the TIMER has resolution 32bits) multiplied by 4 and then returned into a uint64_t container.

So I believe normal modulo arithmetic cannot apply here, and special handling of the overflow would be needed rather than just subtracting:
```
return (*end - *start);
```

This is my quick and dirty solution when returning delta:
```
....
uint32_t delta = ((uint32_t)*end) - ((uint32_t)*start);
return delta;
```
Can you please confirm me if this does make sense to you or the problem is somewhere else?

**To Reproduce**
I discovered this by using a routines measuring an average of execution over time, as the problem is occurring only in the specific overflow cases.

**Expected behavior**
Handling of the timer count overflow

**Impact**
Annoyance when tracking time.

**Environment**
Zephyr v2.6.0

davidelogi, 2021-10-04:
Is there any maintainer that can have a quick look into this?
Happy to open a proper PR if you just confirm my findings

nordic-krch, 2021-10-04:
@davidelogi PR is opened. See #38805.

closed 2021-12-18 by github-actions[bot]