zephyrproject-rtos/zephyr #37391: 4 Bits of IQ Samples Are Removed (Direction Finding Based on CTE)

saleh-unikie, 2021-08-03:
**Describe the bug**
In "Direction Finding Examples", the reported IQ samples are shifted 4 bits after the measurement (4 bits are lost). So reported samples to the application is 8 bits instead of 12 bits.
I can not find any reason behind this shift operation which removes a part of data which is important and usable by the application.

The shift operation is implemented here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/hci/hci.c#L2681
and
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/hci/hci.c#L2683

Reference:
You can find here a short description of IQ sampling that implicitly declares the samples are 12 bit signed. It doesn't mention anything about removing lower bits in order to get a useful data:
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf5340%2Fradio.html&anchor=concept_dfe_iq_sampling 

Direction Finding Examples:
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/direction_finding_connectionless_rx
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/direction_finding_connectionless_tx

**To Reproduce**
Run the "direction_finding_connectionless_tx" and "direction_finding_connectionless_rx" examples and print the generated IQ samples.

**Expected behavior**
The IQ samples should be 12 bits while they are 8 bit.

**Impact**
The accuracy of the IQ samples has decreased which prevents the application to calculate the angle and direction of the devices correctly.

**Logs and console output**
N/A

**Environment (please complete the following information):**
-     OS: Linux
-     Toolchain: nRF Connect SDK
-     v2.6.0

**Additional context**
N/A

saleh-unikie, 2021-08-03:
A fix has provided: https://github.com/zephyrproject-rtos/zephyr/pull/37411

ppryga, 2021-08-04:
@saleh-unikie The IQ samples are stored in single byte due to Bluetooth Core spec. Please take a look into BT Core spec 5.1 Vol 4. Part E, 7.7.65.21 LE Connectionless IQ Report event.

> I_Sample[i]: Size: Sample_Count × 1 octet
> Q_Sample[i]: Size: Sample Count × 1 octet.

This is the reason for dropping bits from original value.

closed 2021-08-04 by ppryga