zephyrproject-rtos/zephyr #37412: IQ samples are not correct during the "reference period" of CTE signal

saleh-unikie, 2021-08-03:
**Describe the bug**
I am working on Bluetooth direction finding examples as can be found here:
[https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/direction_finding_connectionless_rx](url)
[https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/direction_finding_connectionless_tx](url)

I have compared the result with a third party product which has implemented similar method based on nRF52833 chipsets and is working properly. I found the reported IQ samples which are accessible in the receiver side, are not correct at least during the "reference period" of CTE signal.

To print the IQ samples I am using this code at the main.c of the receiver side:
```
static void cte_recv_cb(struct bt_le_per_adv_sync *sync,
			struct bt_df_per_adv_sync_iq_samples_report const *report)
{
        for(int i=0; i<report->sample_count; i++){
		printk(",%d,%d", report->sample[i].i, report->sample[i].q);
	}
        printk("\n");
}
```
Then I have plotted the received samples for both 1us and 2us switching slots:
This plot belongs to 1us switching slot (using the PCB antenna):
[image]

This plot is IQ samples by 2us switching slots (using the PCB antenna):
[image]

IQ samples with 1us switching slots and using array antenna:
[image]

And it is the results of the third party product which has implemented the similar methods based on CTE signals (using the antenna array):
[image]

As you can see in the above images, I and Q signals during the "reference period" has different phases in the third party product but they have same phases in the examples available in Zephyr project. As described in the below white paper, they should have different phases, so some things is wrong with this direction finding examples.
White paper: [https://infocenter.nordicsemi.com/pdf/nwp_036.pdf](url) (page 10)

Please note all above images are based on captured data while devices were operating in AoA mode.

**To Reproduce**
Run the "direction_finding_connectionless_tx" and "direction_finding_connectionless_rx" examples and print the generated IQ samples. Then plot the I and Q samples using a visualized tool. (I have used pyplot by python language)

**Expected behavior**
The "I" and "Q" samples should be in different phases during the "reference period" of CTE signal while they are not.

**Impact**
It is not possible to calculate the angle of AoA tag which is the main application of above examples to determine the location of a tag. 

**Logs and console output**
N/A

**Environment (please complete the following information):**
OS: Linux
Toolchain: nRF Connect SDK
v2.6.0

**Additional context**
For more information about sampling during the "reference period" of CTE signal refer to this white paper:
[https://infocenter.nordicsemi.com/pdf/nwp_036.pdf](url) (page 11)

In total, 8 samples are captured during this period and it is not dependent on the antenna type, so in all above conditions, it should have similar behavior despite of using PCB antenna or antenna array.

saleh-unikie, 2021-08-04:
UPDATE:  I executed the third party implementation (which our algorithm can calculate the transmitter angle based on its results properly) with the pcb antenna on nRF52833-DK (same as what I have used to run Zephyr examples in two first images above) and you can see the results here:
[image]

It means using the "antenna array" or "pcb antenna" doesn't affect the results of the samples during the "Reference Period" of CTE signals. The reason is that for sampling the reference period, same antenna has been used for all 8 samples during this period.

saleh-unikie, 2021-08-04:
Would you please check this issue? @ppryga

saleh-unikie, 2021-08-04:
Thank you @ppryga to handle this issue :)
I guess the problem could be a wrong frequency of CTE signal. Also I have opened a ticket in Nordic devzone corresponding this problem: https://devzone.nordicsemi.com/f/nordic-q-a/78189/problem-with-i-and-q-samples-during-the-reference-period-of-cte-packets

saleh-unikie, 2021-08-04:
I think the problem is frequency of CTE signal which is transmitted by the AoA tag. As mentioned in these resources, the CTE signal should carry a 250kHz frequency:

Ref1: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fnwp_036%2FWP%2Fnwp_036%2Fconstant_tone_extension.html
Ref2: https://dev.ti.com/tirex/explore/node?node=AHYhhuDNTaRXzkOlahOlvA__pTTHBmu__LATEST
Ref3: https://www.silabs.com/documents/public/application-notes/an1297-custom-direction-finding-solutions-silicon-labs-bluetooth.pdf

It means the wave period is 4us, thus two cycle should be visible during the "Reference Period", but seems there are more than 2 cycle at the "Reference Period". Maybe the transmitter is sending an incorrect signal as CTE.

ppryga, 2021-08-06:
The issues was answered in the Devzone. There is nothing to do about it in the Zephyrs code. Observed lack of shift in IQ data is related with sampling spacing. For more details please do to Devzone: https://devzone.nordicsemi.com/f/nordic-q-a/78189/problem-with-i-and-q-samples-during-the-reference-period-of-cte-packets

closed 2021-08-06 by ppryga