zephyrproject-rtos/zephyr #37429: NRF52840 hangs in z_add_timeout function

gpuzio, 2021-08-04:
**Describe the bug**
Nrf52840 (on Thingy91) randomly hangs in z_add_timeout() function. Most probably it happens only when Bluetooth is enabled and scanning is running on nrf52840. 

**What have you tried to diagnose or workaround this issue?**

I noticed that NRF52840 is facing occasional, very random watchdog resets. In my application watchdog is feed from main loop (reset happen after 1 minute). I started investigation what is reason for these resets, I disabled watchdog and connected debugger. I have following observations:

- with watchdog disabled hangs still happens, it last around 9 minutes and after this everything went to normal state.
- During that 9 minutes I don’t see any logs, which means that interrupts are not working
- Based on timestamps in logs I can say, that hangs always last the same amount of time – in my case break in logs is always 9 minutes and 2 seconds
- Hangs happen very randomly – sometimes it is not seen for several hours, sometimes it happens in 10 minutes after reset.
- When program execution is stopped with debugger it is always pointing to z_add_timeout() where infinite for loop is executed:

```
0x0003d9da in z_add_timeout (to=0x200039f0 <z_idle_threads+24>, fn=fn@entry=0x48011 <z_thread_timeout>, timeout=...) at /home/kazik/nordic/sdk/ncs/zephyr/include/sys/dlist.h:341
341		return (node != NULL) ? sys_dlist_peek_next_no_check(list, node) : NULL;
(gdb) next
108				if (t->dticks > to->dticks) {
(gdb) next
113				to->dticks -= t->dticks;
(gdb) next
107			for (t = first(); t != NULL; t = next(t)) {
(gdb) next
108				if (t->dticks > to->dticks) {
(gdb) next
113				to->dticks -= t->dticks;
(gdb) next
107			for (t = first(); t != NULL; t = next(t)) {
(gdb) next
108				if (t->dticks > to->dticks) {
(gdb) next
113				to->dticks -= t->dticks;

(gdb) print *to
$2 = {node = {{head = 0x200039f0 <z_idle_threads+24>, next = 0x200039f0 <z_idle_threads+24>}, {tail = 0x200039f0 <z_idle_threads+24>, prev = 0x200039f0 <z_idle_threads+24>}}, 
  fn = 0x48011 <z_thread_timeout>, dticks = 0}
(gdb) print *t
$3 = {node = {{head = 0x200039f0 <z_idle_threads+24>, next = 0x200039f0 <z_idle_threads+24>}, {tail = 0x200039f0 <z_idle_threads+24>, prev = 0x200039f0 <z_idle_threads+24>}}, 
  fn = 0x48011 <z_thread_timeout>, dticks = 0}
```

It looks that node is single element in timeout_list and have dticks = 0 and also the same element is being passed to z_add_timeout() function and for loop can’t be break. I don't understand how this situation happened and how it normally ends. I suspect that different timeout needs to be added to break this loop. When program execution is stopped with debugger break will never happen (even I let program to continue execution) – I also don’t know why this happen (probably Bluetooth module stop working correctly)

**To Reproduce**
Enable Bluetooth, start scanning and wait

**Expected behavior**
Hangs in z_add_timeout does not happened 

**Impact**
showstopper

**Logs and console output**
Logs from gdb above

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain nrfConnect SDK (NCS version 1.4.2)
 - Commit SHA  (e34b2f477e1adc26b66690171508bfa102eb8538, tag: v2.4.0-ncs2-rc1, tag: v2.4.0-ncs2)

**Additional context**
With new NCS (version 1.6.1) it looks that problem  is gone, however switching to other version is not preferred option.

nordic-krch, 2021-08-04:
You are probably using Softdevice Link Layer. Can you check if issue happens also when zephyr LL is used (`CONFIG_BT_LL_SW_SPLIT=y`)

gpuzio, 2021-08-04:
Yes, I'musing Softdevice Link Layer, I have switched to zephyr LL few hours ago and I'm testing it right now, so far no issue seen. I will update modified firmware on multiple devices so I should have results in 2 days (or earlier if problem appear).

Is it known issue with Softdevice LL ?

carlescufi, 2021-08-05:
FYI @rugeGerritsen @seko-nordic

nordic-krch, 2021-08-05:
there might be two things caused by softdevice LL using ZLI (non-maskable interrupts)
- from ZLI interrupt context kernel api must not be called but is called (e.g. inside ISR_DIRECT_PM`)
- timer driver is somehow failing because it is not ready to be interrupted by ZLI while being in lock.

Regarding first option, those are places where this happens:
https://github.com/nrfconnect/sdk-nrf/blob/41202f304f0e67423448afc5368fd4d640cb73e6/subsys/mpsl/mpsl_init.c#L82
https://github.com/nrfconnect/sdk-nrf/blob/41202f304f0e67423448afc5368fd4d640cb73e6/subsys/mpsl/mpsl_init.c#L95
https://github.com/nrfconnect/sdk-nrf/blob/41202f304f0e67423448afc5368fd4d640cb73e6/subsys/mpsl/mpsl_init.c#L107

Can you try to remove calls to `ISR_DIRECT_PM` calls from those irq handlers and switch back to Softdevice LL?

gpuzio, 2021-08-06:
Sure, I will check this today and leave one device for test during weekend. 

Regarding Zephyr Link Layer (CONFIG_BT_LL_SW_SPLIT=y) - it looks that this completely resolved problem with hanging (and watchdog resets) on my devices. It is running on couple of devices for 2 days without single reset. I noticed only one problem when I switched to Zephyr LL - bt_scan_blocklist_clear function doesn’t clear blocked devices until bt_scan_stop/bt_scan_start is called – with Softdevice LL this worked, but I didn’t investigate this problem, just add stop/start scanning.

gpuzio, 2021-08-07:
I can say now that removing calls to ISR_DIRECT_PM did't resolve problem with resets.

carlescufi, 2021-08-12:
@nordic-krch how should we proceed here you think?

carlescufi, 2021-08-12:
@gpuzio could you try enabling ZLI in the open source controller to see if that reproduces the issue as well? You can enable it with [this config option](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_BT_CTLR_ZLI.html#std-kconfig-CONFIG_BT_CTLR_ZLI), and you need to enable [this option](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_ZERO_LATENCY_IRQS.html#std-kconfig-CONFIG_ZERO_LATENCY_IRQS) and [this one](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_BT_CTLR_ADVANCED_FEATURES.html#std-kconfig-CONFIG_BT_CTLR_ADVANCED_FEATURES)

scath999, 2023-02-20:
This issue seems to have appeared on Laird's MG100 when attempting to run the `http_get` example.

Looking for guidance on how to resolve this.

Note: As a shot in the dark, I've tried setting `CONFIG_BT_LL_SW_SPLIT=y` in `prj.conf` - this causes a bus fault:

```

[00:00:04.514,434] <err> os: ***** BUS FAULT *****
[00:00:04.514,434] <err> os:   Instruction bus error
[00:00:04.514,434] <err> os: r0/a1:  0x00000000  r1/a2:  0x2002665c  r2/a3:  0x00000004
[00:00:04.514,465] <err> os: r3/a4:  0x94f7c0b5 r12/ip:  0x00000000 r14/lr:  0x0003146d
[00:00:04.514,465] <err> os:  xpsr:  0x21000000
[00:00:04.514,465] <err> os: Faulting instruction address (r15/pc): 0x94f7c0b4
[00:00:04.514,465] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:04.514,495] <err> os: Current thread: 0x2000f3e4 (main)
[00:00:04.665,557] <err> fatal_error: Resetting system
```

Please let me know what more diagnostic information I can provide.

**OS:** Win10
**Board:** nrf52840 on Laird MG100 (Note: I'm using the actual device, not the MG100 DVK)
**Modem:** hl8700
**Zephyr Version:** 2.3.0 (note, this is the version included with Laird's development firmware - I have not yet upgraded this, and am unsure if I should).
**JLink:** nrf52840 DK
**Sample Package:** [Laird MG100 LTE-M/AWS](https://github.com/LairdCP/MG100_firmware/blob/master/docs/readme_ltem_aws.md), with code from Zephyr `http_get` sample project appended and altered as appropriate.

**Expected Result:**
1) A connection between the MG100 and remote host over HTTP(S) should be established,
2) A GET request should be sent by the MG100 to the remote host using demonstration code.
3) A properly formed response should be received by the MG100 from the remote host with the proper payload.

**Actual Result:**
1) The connection and send requests execute (using `connect()` and `send()` methods demonstrated in `http_get/src/main.c : lines 109-110`.
2) The methods called by these factory calls _do complete,_ but execution is never returned to `main`.
- NOTE: The connection is never actually established - I see no record of any connection between the MG100 and my server).
3) When debugging via GDB, it is observed that the execution path enters `z_add_timeout()` and never leaves. 
- Specifically, execution enters the `for (t = first(); t != NULL; t = next(t))` block of code between lines 111-120 in `timeout.c` and never leaves.
4) The MG100 continues to execute other threads. As mentioned in point 2, no connection is ever established and no send is ever performed.

**Other sources consulted:**
I have looked at the [issue ](https://github.com/zephyrproject-rtos/zephyr/issues/9944)regarding `connect` not sending a new SYN request in the case where a first connection attempt fails. The workaround did not help.

cvinayak, 2024-03-07:
@scath999 You comment may have been overlooked... is this still an issue, if so, could you create a new bug report?