zephyrproject-rtos/zephyr #37436: Delayed startup due to printing over not ready console

emob-nordic, 2021-08-04:
I'm posting this report, because i believe there are two things that should be reviewed and fixed by the maintainers. I already got the fix but i would like to see your opinion on this (@jfischer-no and @nordic-krch).

**Describe the bug**
Startup of the main application is delayed by up to 5s, when console is enabled on USB CDC ACM device.
The issue was spotted in NCS sample for Thingy53 board, however i believe the fix should land in Zephyr.

**To Reproduce**
1. apply the patch. This is done just to better visualise the issue. All it does it lights up Thingy53's rgb LED at the beggining of the `main` function, when dk_library is initialized.
``` patch
diff --git a/samples/bluetooth/peripheral_lbs/boards/thingy53_nrf5340_cpuapp.conf b/samples/bluetooth/peripheral_lbs/boards/thingy53_nrf5340_cpuapp.conf
index a83ea90fc..227092293 100644
--- a/samples/bluetooth/peripheral_lbs/boards/thingy53_nrf5340_cpuapp.conf
+++ b/samples/bluetooth/peripheral_lbs/boards/thingy53_nrf5340_cpuapp.conf
@@ -17,7 +17,7 @@ CONFIG_BT_L2CAP_TX_MTU=260
 CONFIG_BT_BUF_ACL_RX_SIZE=264
 
 # LEDs are not inverted on Thingy:53
-CONFIG_DK_LIBRARY_INVERT_LEDS=n
+CONFIG_DK_LIBRARY_INVERT_LEDS=y
 
 # Use deferred logging
 CONFIG_NCS_SAMPLES_DEFAULTS=n
```
2. `west build -b thingy53_nrf5340_cpuapp`
3. `west flash`
4. See that rgb LED is white after up to 5 seconds from finished flasing. Reset the board few times.

I've already fixed the bug by adding `CONFIG_LOG_PRINTK=y` to the sample:
https://github.com/nrfconnect/sdk-nrf/pull/5230

**Additional context**
There are two things i would like to talk through:
1. I think `CONFIG_LOG_PRINTK=y` should be defaulted to `y` if USB console backed is chosen. Similarly as it was done here:
https://github.com/zephyrproject-rtos/zephyr/pull/14465, in response to issue: https://github.com/zephyrproject-rtos/zephyr/issues/11393.

@nordic-krch what do you think?

2. When i were trying to find the bug i spotted that the console is calling `poll_out` to send the character using cdc_acm backend. Implementation there uses a semaphore with 100ms waiting time.
```
static void cdc_acm_poll_out(const struct device *dev,
				      unsigned char c)
{
	struct cdc_acm_dev_data_t * const dev_data = DEV_DATA(dev);

	cdc_acm_fifo_fill(dev, &c, 1);
	k_sem_take(&dev_data->poll_wait_sem, K_MSEC(100));
}
```
The semaphore would not be available until the USB device enumerates and thus it hangs here for each character in the printed message.
My question here is. Is this `100ms` still relevant? It comes from 1st commit to cdc_acm class and for me it doesn't make any sense to wait for this semaphore here. I think the code should look more like:
```
static void cdc_acm_poll_out(const struct device *dev,
				      unsigned char c)
{
	struct cdc_acm_dev_data_t * const dev_data = DEV_DATA(dev);

        if (k_sem_take(&dev_data->poll_wait_sem, K_NO_WAIT)) {
		cdc_acm_fifo_fill(dev, &c, 1);
        } else if (!IS_ENABLED(CONFIG_USB_UART_CONSOLE)) {
                /* Todo: check on which cdc_acm device the console is used. */
                LOG_ERR("Dropped character");
        }
}
```
@jfischer-no What's your opinion on this, am i missing something important?

Another thing for usb part is that mentioned semaphore is given on `USB_DC_RESET` and `USB_DC_DISCONNECTED` which i believe is not correct. This semaphore should only be given when the device is configured or resumed from suspend when previously configured.
If we leave this as is and apply `LOG_PRINTK` fix we will probably see some of those:
```
static int cdc_acm_fifo_fill(const struct device *dev,
			     const uint8_t *tx_data, int len)
{
	struct cdc_acm_dev_data_t * const dev_data = DEV_DATA(dev);
	size_t wrote;

	LOG_DBG("dev_data %p len %d tx_ringbuf space %u",
		dev_data, len, ring_buf_space_get(dev_data->tx_ringbuf));

	if (!dev_data->configured || dev_data->suspended) {
		LOG_WRN("Device not configured or suspended, drop %d bytes",
			len); // <==== those
		return 0;
	}
```
One last thing is why is mentioned semaphore initialized to `K_SEM_MAX_LIMIT`. Shouldn't this be something closer to:
`MAX(USB_BULK_EP_SIZE, COND_CODE_1(COFNIG_CONSOLE, CONSOLE_MAX_MESSAGE_SIZE, 0))?` - pseudo code.

jfischer-no, 2021-08-04:
>     1. I think `CONFIG_LOG_PRINTK=y` should be defaulted to `y` if USB console backed is chosen. Similarly as it was done here:

NACK for that, it should not be default because there is no issue with the back-end self. Rather the other way round, printk should not be used with CDC ACM UART driver without wait-for-DTR.

emob-nordic, 2021-08-04:
How can we assure no printk() were used on system boot up? The problem is there are plenty of places where `LOG` or `printk` is used to log some information during system boot up, those will trigger described delay if console is cdc acm.

carlescufi, 2021-09-08:
@emob-nordic can we close this now?

emob-nordic, 2021-09-08:
Resolved by: #37503

closed 2021-09-08 by emob-nordic