zephyrproject-rtos/zephyr #37441: Native POSIX Flash Storage Does not Support Multiple Instances

bob2oneil, 2021-08-04:
**Describe the bug**
Flash emulation under Native-POSIX creates a file as "flash.bin" that lives within the same folder as the binary.  When considering multiple instances of a Zephyr binary, running in a network emulation environment, this single file can not be used exclusively for each instance.

For example, if settings are saved to flash.bin for one instance, those unique per instance settings are then somewhat forced on the other instances.

A more elegant solution would be to have a kernel configuration setting that allows each instance to specify the base folder where the flash.bin file should reside.  This is similar to the way that there is a kernel configuration setting to specify the base name for virtual Ethernet interfaces (defaults to "zeth", but can be customized to support multiple instances as e.g. "zeth1.", "zeth2.", etc).

An example might be a default as say "etc/zeth/1" for Linux deployment, or perhaps 'tmp/1" for more platform neutral use.

.
What have you tried to diagnose or workaround this issue?

A workaround is to copy each Zephyr binary into it's own folder for execution.

**To Reproduce**
Steps to reproduce the behavior:
1. Create an application using flash/littlefs support
2. Make for Native-POSIX board target
3. make
4. See error

**Expected behavior**
Separation of settings per instance that does not require physical separation of the Zephyr binary.

**Impact**
Annoyance

**Logs and console output**
Not applicable

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

de-nordic, 2021-08-05:
This is not a bug. It has been designed that way and, probably, some CI scripts are hardcoded to search for that flash.bin.
I think the issue should be re-classified as enhancement/feature request.

bob2oneil, 2021-08-05:
I am fine with a reclassification, but in  practical sense, it requires multiple instances of the Zephyr binary to be located in different folders to be able to have unique persistent flash storage.

Laczen, 2021-08-07:
@bob2oneil, I am not completely sure that I understand your problem correct. You know it is possible to pass the flash.bin file to zephyr.exe ?

bob2oneil, 2021-08-07:
I was not aware of that, thanks.  I will check that this argument is provided when flash is enabled in a build.

closed 2021-08-07 by bob2oneil