zephyrproject-rtos/zephyr #37446: Sensor driver: ST LPS22HH undeclared functions and variables

rajeev1986, 2021-08-04:
While going over the LPS22HH sensor driver, I noticed that there is a function call made which is not declared anywhere in the associated files.
In [lps22hh_init_chip()](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/lps22hh/lps22hh.c#L148), -> ` lps22hh_device_id_get()` is called which is undeclared. The compiler should have been able to pick this up.
Secondly, a variable `LPS22HH_ID` is not declared or assigned a value anywhere while it is used to compare the validity of the sensor `chip_id` resulting in an invalid chip.

Thanks

FRASTM, 2021-08-05:
however the **lps22hh_device_id_get()** exists in the 
https://github.com/zephyrproject-rtos/hal_st/tree/master/sensor/stmemsc/lps22hh_STdC/driver/lps22hh_reg.c

the **LPS22HH_ID** is defined in the https://github.com/zephyrproject-rtos/hal_st/tree/master/sensor/stmemsc/lps22hh_STdC/driver/lps22hh_reg.h
```
/** Device Identification (Who am I) **/
#define LPS22HH_ID                              0xB3U
```

gmarull, 2021-08-05:
Sample just builds fine with boards that have the sensor: `west build -b sensortile_box samples/sensor/lps22hh`.

Running:
```
west build -b sensortile_box samples/sensor/lps22hh/ -- -DCMAKE_EXPORT_COMPILE_COMMANDS=ON 
```

And using letting VSCode use the generated compile_commands.json allows me to click and resolve that function.

Closing the issue as this is not a bug.

closed 2021-08-05 by gmarull

rajeev1986, 2021-08-05:
Thanks for the clarification. Much appreciated.