zephyrproject-rtos/zephyr #37460: tests/kernel/sched/schedule_api/kernel.scheduler and tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout failed on nsim_hs_smp board

IRISZZW, 2021-08-05:
**Describe the bug**
tests/kernel/sched/schedule_api/kernel.scheduler and tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout failed on nsim_hs_smp board

**To Reproduce**
Steps to reproduce the behavior:
1. ./scripts/twister -p nsim_hs_smp -O build.sanity.nsim_hs_smp -n -v -T tests/kernel/sched/schedule_api -T tests/kernel/fifo/fifo_timeout
2. See error

**Logs and console output**
```
./scripts/twister -p nsim_hs_smp -O build.sanity.nsim_hs_smp -n -v -T tests/kernel/sched/schedule_api -T tests/kernel/fifo/fifo_timeoutKeeping artifacts untouched
INFO    - Zephyr version: zephyr-v2.6.0-1753-g365ff6db9f02
INFO    - JOBS: 12
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 6 test scenarios (6 configurations) selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/6 nsim_hs_smp               tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout FAILED Failed (nsim 4.480s)
ERROR   - see: /SCRATCH2/Watson/zephyrproject/zephyr/build.sanity.nsim_hs_smp/nsim_hs_smp/tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout/handler.log
INFO    - 2/6 nsim_hs_smp               tests/kernel/sched/schedule_api/kernel.scheduler.multiq_no_timeslicing PASSED (nsim 3.978s)
INFO    - 3/6 nsim_hs_smp               tests/kernel/sched/schedule_api/kernel.scheduler.dumb_no_timeslicing PASSED (nsim 4.254s)
INFO    - 4/6 nsim_hs_smp               tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing PASSED (nsim 4.179s)
INFO    - 5/6 nsim_hs_smp               tests/kernel/sched/schedule_api/kernel.scheduler   FAILED Failed (nsim 4.446s)
ERROR   - see: /SCRATCH2/Watson/zephyrproject/zephyr/build.sanity.nsim_hs_smp/nsim_hs_smp/tests/kernel/sched/schedule_api/kernel.scheduler/handler.log
INFO    - 6/6 nsim_hs_smp               tests/kernel/sched/schedule_api/kernel.scheduler.multiq PASSED (nsim 6.142s)

INFO    - 4 of 6 test configurations passed (66.67%), 2 failed, 0 skipped with 0 warnings in 30.96 seconds
INFO    - In total 106 test cases were executed, 40 skipped on 1 out of total 383 platforms (0.26%)
INFO    - 6 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing xunit report /SCRATCH2/Watson/zephyrproject/zephyr/build.sanity.nsim_hs_smp/twister.xml...
INFO    - Writing xunit report /SCRATCH2/Watson/zephyrproject/zephyr/build.sanity.nsim_hs_smp/twister_report.xml...
INFO    - Run completed
```
tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout/handler.log:
```
-- west flash: using runner mdb-nsim
Console now belongs to UART, hit CRTL-] to return to simulator.
*** Booting Zephyr OS build zephyr-v2.6.0-1753-g365ff6db9f02  ***
Running test suite test_fifo_timeout
===================================================================
START - test_timeout_empty_fifo
 PASS - test_timeout_empty_fifo in 0.46 seconds
===================================================================
START - test_timeout_non_empty_fifo
 PASS - test_timeout_non_empty_fifo in 0.9 seconds
===================================================================
START - test_timeout_fifo_thread
 PASS - test_timeout_fifo_thread in 0.70 seconds
===================================================================
START - test_timeout_threads_pend_on_fifo
 thread (q order: 2, t/o: 0, fifo 0x80000000)
 thread (q order: 0, t/o: 20, fifo 0x80000000)
 thread (q order: 3, t/o: 10, fifo 0x80000000)
 thread (q order: 1, t/o: 40, fifo 0x80000000)
 thread (q order: 4, t/o: 30, fifo 0x80000000)
 PASS - test_timeout_threads_pend_on_fifo in 0.123 seconds
===================================================================
START - test_timeout_threads_pend_on_dual_fifos
 thread (q order: 0, t/o: 0, fifo 0x80000018)
FAIL - test_multiple_threads_pending@186.  *** thread 3 woke up, expected 1

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/fifo/fifo_timeout/src/main.c:467: test_timeout_threads_pend_on_dual_fifos: rv not equal to TC_PASS

 FAIL - test_timeout_threads_pend_on_dual_fifos in 0.157 seconds
===================================================================
START - test_timeout_threads_pend_fail_on_fifo
FAIL - test_multiple_threads_get_data@239.  *** thread 5 woke up, expected 0

[... 127 lines cut ...]
START - test_k_thread_priority_set_init_null
 SKIP - test_k_thread_priority_set_init_null in 0.5 seconds
===================================================================
START - test_k_thread_priority_set_overmax
 SKIP - test_k_thread_priority_set_overmax in 0.5 seconds
===================================================================
START - test_k_thread_priority_set_upgrade
 SKIP - test_k_thread_priority_set_upgrade in 0.5 seconds
===================================================================
START - test_k_wakeup_init_null
 SKIP - test_k_wakeup_init_null in 0.5 seconds
===================================================================
Test suite threads_scheduling failed.
===================================================================
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK)
 - Commit SHA: 365ff6db9f02fce0f9d1711590ab55e79c8ca950

**Additional context**
I use git bisect found these failures were introduced by this commit: https://github.com/zephyrproject-rtos/zephyr/pull/36856/commits/00f95032938e1387126f453b64f8c479987ca16e

IRISZZW, 2021-08-05:
if I add `__attribute_const__` for `z_current_get`, it works fine.
```
diff --git a/include/kernel.h b/include/kernel.h
index 4c2aea5..1b3b66c 100644
--- a/include/kernel.h
+++ b/include/kernel.h
@@ -494,7 +494,7 @@ __syscall void k_wakeup(k_tid_t thread);
  *
  * @return ID of current thread.
  */
-__syscall k_tid_t z_current_get(void);
+__syscall k_tid_t z_current_get(void) __attribute_const__;
 
 #ifdef CONFIG_THREAD_LOCAL_STORAGE
 /* Thread-local cache of current thread ID, set in z_thread_entry() */
```

IRISZZW, 2021-08-09:
@dcpleung  Could you please look at this issue?

dcpleung, 2021-08-09:
I could add `__attribute_const__`. But this seems to be a deeper issue as slowing down a few functions calls should not result in those tests failing.

PR #37553 (2021-08-18): kernel: mark z_current_get() as const
In commit 00f95032938e1387126f453b64f8c479987ca16e to make
k_current_get() work without syscall, a new z_current_get()
was introduced since there are times when thread local
storage has not been initialized. However, this was not
marked with const attribute the same as k_current_get().
This may result in slower compiled code as each call to
z_current_get() may actually need to go through the whole
function call process instead of reusing the result from
previous call in the same scope. So add const attribute
to z_current_get() to restore the old behavior.
 [... cut]
commit 8530cfa7 (2021-08-18): kernel: mark z_current_get() as const

In commit 00f95032938e1387126f453b64f8c479987ca16e to make
k_current_get() work without syscall, a new z_current_get()
was introduced since there are times when thread local
storage has not been initialized. However, this was not
marked with const attribute the same as k_current_get().
This may result in slower compiled code as each call to
z_current_get() may actually need to go through the whole
function call process instead of reusing the result from
previous call in the same scope. So add const attribute
to z_current_get() to restore the old behavior. [... cut]
closed 2021-08-18 by cfriedt