zephyrproject-rtos/zephyr #37461: Schedule of LE audio in zephyr

xin10pan, 2021-08-05:
Hi，
For LE Audio， Zephyr now supports ISO and MICP and VCP related profiles.  Is there a plan to support BAP&MCP&CCP？Is there a schedule for LE audio development in Zephyr?
Looking forward to your reply!

carlescufi, 2021-08-05:
Hi @xin10pan. Some of the work implementing those services is ongoing in a private repository because some of the the services you mention (BAP in particular) are not public yet. If you'd like to get access to that repo you would need to join the Zephyr Project as a company. Otherwise you can wait until all of that work trickles down to the public repo.
Regarding LC3, we have a preliminary codec here: https://github.com/zephyrproject-rtos/liblc3codec but more work is expected in the future.

closed 2021-08-05 by carlescufi

xin10pan, 2021-08-06:
When will these profiles be pushed to public repo? Is there a schedule?