zephyrproject-rtos/zephyr #37462: Bluetooth: Advertising becomes scannable even if BT_LE_ADV_OPT_FORCE_NAME_IN_AD is set

Thalley, 2021-08-05:
**Describe the bug**
When creating an extended advertiser or just using the `bt_le_adv_start` with the `BT_LE_ADV_OPT_USE_NAME` option, it implicitly makes the advertisements scannable, as it would put the name in the scan response data. 

However, we introduced another option, `BT_LE_ADV_OPT_FORCE_NAME_IN_AD`, which forces the name to be in the advertising data, but the advertisement will still become scannable, even if the upper layer didn't request it to be. 

**To Reproduce**
N/A

**Expected behavior**
If `BT_LE_ADV_OPT_FORCE_NAME_IN_AD` is set with `BT_LE_ADV_OPT_USE_NAME`, `BT_LE_ADV_OPT_SCANNABLE` isn't set and no scan response data was provided, then the advertisement should not be scannable.

This can be fixed by changing how we determine whether it should be scannable from doing 
```
scannable = param->options & BT_LE_ADV_OPT_USE_NAME;
```
to
```
scannable = (param->options & BT_LE_ADV_OPT_USE_NAME != 0) && (param->options & BT_LE_ADV_OPT_FORCE_NAME_IN_AD == 0);
```

**Impact**
Minor, will just making it scannable without any scan response data. 

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - Commit SHA or Version used: 1c0ed946018e4f1ee4b766d5db811d097f44f2ba

**Additional context**
N/A

carlescufi, 2021-08-12:
@lairdjm any comments on this one?

lairdjm, 2021-08-12:
Looks good to me

PR #37673 (2021-08-20): Bluetooth: host: Fix non-connectable advertiser configured as scannable
Fix non-connectable advertiser configured as ADV_SCAN_IND when configured
by application to have the device name appear in the advertising data
instead of the scan response data.

Disallow creating an extended scannable advertiser with the device name
configured to appear in the advertising data.
This would fail in either the call to bt_le_ext_adv_start or
bt_le_ext_adv_set_data when the host would try to set advertising data in the
controller.
Instead this now fails in the bt_le_ext_adv_create call.

Document the behavior [... cut]
closed 2021-08-20 by carlescufi