zephyrproject-rtos/zephyr #37472: Corrupted timeout on poll for offloaded sockets

emillindq, 2021-08-05:
**Describe the bug**
When pulling latest Zephyr for our project, I noticed that `poll()` suddenly went from blocking for 10ms to 30000ms. It seems like [this PR](https://github.com/zephyrproject-rtos/zephyr/pull/36220) is causing this timeout corruption with this addition:
https://github.com/zephyrproject-rtos/zephyr/blob/ca5921845d4241ce9da9496a8e97d7dd6c1c4bae/subsys/net/lib/sockets/sockets.c#L1455-L1462

It seems like the intention is to convert ticks to ms, but the function `sys_clock_timeout_end_calc` returns **at which tick in the future the timeout will happen**. For the argument K_NO_WAIT it simply returns `sys_clock_tick_get()` with the definition
> Return the current system tick count
So instead of waiting 10ms, it waits for as long as the program has been running previously, plus 10ms, is my interpretation.

I'm not sure if this was the intention, but further down `poll_timeout` is used as time in *ms* as argument for `k_sem_take()`, at least in 
https://github.com/zephyrproject-rtos/zephyr/blob/ca5921845d4241ce9da9496a8e97d7dd6c1c4bae/drivers/modem/modem_socket.c#L307

Hence why it blocks for 30000ms.

Is my conclusion correct?

**To Reproduce**
Steps to reproduce the behavior:
1. Create offloaded socket
2. Run the program for a while
3. poll(..., ..., 10) and it will block  for as long as it has run

**Expected behavior**
Poll timeout should be intact when passed down

**Impact**
Showstopper

**Logs and console output**
N/A but it returns from poll() after 30s

**Environment (please complete the following information):**
 - Commit SHA 997bbcb000c35f1d0de90668bb824bcce594daed
- SARA-R4 is the offloaded socket, and it is used behind a DTLS socket

emillindq, 2021-08-05:
If changing to
```
		int poll_timeout;

		if (K_TIMEOUT_EQ(timeout, K_FOREVER)) {
			poll_timeout = SYS_FOREVER_MS;
		} else {
			poll_timeout = k_ticks_to_ms_floor32(
				timeout.ticks);
		}
```
it waits for 10ms again

rlubos, 2021-08-06:
Indeed, the `k_timeout_t` to ms conversion doesn't seem right in this case and was somehow overlooked in the original PR. I think that your proposal looks right, feel free to submit a PR!

emillindq, 2021-08-06:
PR created, our unit tests pass with the fix

PR #37496 (2021-08-06): net: socket: fix regression causing corrupted poll timeout
k_timeout_t was converted to ticks using a nonsense function
causing poll timeout corruption for offloaded sockets; this
commit uses ticks directly from the struct instead.

Fixes #37472
commit dab4616e (2021-08-06): net: socket: fix regression causing corrupted poll timeout

k_timeout_t was converted to ticks using a nonsense function
causing poll timeout corruption for offloaded sockets; this
commit uses ticks directly from the struct instead.

Fixes #37472
closed 2021-08-06 by cfriedt