zephyrproject-rtos/zephyr #37491: wrong documentation format on DMA peripheral API reference

FRASTM, 2021-08-06:
Not sure if this is the right place to submit:

The Zephyr doc page for the **API Reference / Peripherals / DMA / API Reference** is  not well formatted for the struct `dma_block_config` section.

https://docs.zephyrproject.org/latest/reference/peripherals/dma.html is
[image]

dleach02, 2021-08-10:
@FRASTM can you provide a PR to fix this?

FRASTM, 2021-08-11:
sure, but i have no idea yet of where it is made/stored/pushed. Is there a repo for this ?

dleach02, 2021-08-17:
sorry for the delay. These are extracted from the doxygen style comments in the header files. This particular file, /include/drivers/dma.h, appears to not be specifying the structure parameters or other things (look at dac.h as a reference). When you make the changes, if you want to test locally then rebuild the documents see https://docs.zephyrproject.org/latest/guides/docs/index.html

closed 2021-08-19 by cfriedt