zephyrproject-rtos/zephyr #37495: mcuboot: Doesn't perform revert when the test image is over a certain size.

ycsin, 2021-08-06:
**Describe the bug**

After `hawkbit` downloaded an image and performed a test update, if the device is reset after booted up that test update, for some reason MCUboot doesn't perform swap even though it should, as the image isn't confirmed.

Then if an image is flashed to the `image-0` partition using a hw debugger (Jlink), the MCUboot would think that it needs to `boot_complete_partial_swap`. However, in `boot_swap_image`, the `swap_size` (`copy_size`) from `boot_read_image_size` is likely corrupted?

[image]

And caused it to hard fault while it finds the last sector in the primary slot that needs swapping in `swap_run`

[image]

What's weird is that this happens only when the target image is over certain size, for my case, ~20kb `hello_world` & ~150kb `hawkbit` sample works just fine, as in if reset after the test image is booted, MCUboot will perform a swap.

But later on I tried my own application, which is ~250kB, after reset the MCUboot will not perform a swap, and flashing firmware on top of it will result in a hard fault. I then went on to try the `hawkbit` sample again, but compiled a few shell component to increase the firmware size to > 200kiB,used that as a target image and found that it has the same issue just like my own application.

I wonder what could possibly went wrong that caused the MCUboot to skip swapping for a (large) test image?
And if something can be done in the MCUboot side to prevent a false `boot_complete_partial_swap`?

___

**My board**
- STM32G0B1RE custom board
  - `mcuboot` - 64 kiB
  - `image-0` - 416 kiB 
- 1 MiB external `spi-nor` flash
  - `image-1` - 416 kiB
  - `scratch` - 140 kiB 
- `gsm_ppp` LTE modem

**To Reproduce**
1. Config `hawkbit` so that the final image size is >200 kiB and flash the confirmed image
2. Configure `hawkbit` host to update the device to the unconfirmed image
3. Update and check that it doesn't perform revert
4. Flash the same image over, reset and see hard fault error

**Expected behavior**
It should perform a revert when reset. It shouldn't hard fault when booting an image that is flashed on top after that.

**Impact**
Showstopper

**Logs and console output**
[mcuboot_big-hawkbit_boot.txt](https://github.com/zephyrproject-rtos/zephyr/files/6944909/mcuboot_big-hawkbit_boot.txt)

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: GNUARMEMB
 - v2.6.0-1638-g0296d12675fd
 -

ycsin, 2021-08-17:
@d3zd3z After few days of testing I've edited the title and narrowed the scope of the problem in the description

d3zd3z, 2021-08-23:
I'm not exactly sure I'd call it expected behavior, but it is unsurprising. The easiest fix is going to be either 1. erase the entire partition for the primary image slot before flashing the image with the hw tools, or 2. Pass `--pad` to `imgtool.py` (as well as `--slot-size`), which will fill out the entire image, overwriting whatever state is present.

This could indeed be improved, but it is hard to as there is no correlation between the image in the trailer and a given image itself. If a hw tool replaces the image, but not the trailer, it will still behave. Regardless. Fixing this would be an issue on https://github.com/mcu-tools/mcuboot and not really against Zephyr. Feel free to create an issue there if you'd like.

d3zd3z, 2021-08-23:
I've gone ahead and created #mcu-tools/mcuboot#1116 to track any work on a fix, and I will go ahead and close the issue here on Zephyr.

closed 2021-08-23 by d3zd3z

ycsin, 2021-10-06:
@d3zd3z Hey, give that my image slot has 448kiB:

```dts
...
		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			/*
			 * 448 kB slot-1 for DFU
			 * Must be 4 kB aligned
			 */
			slot1_partition: partition@0 {
				label = "image-1";
				reg = <0x0 0x70000>;
			};
...
```

I've added
```conf
CONFIG_MCUBOOT_EXTRA_IMGTOOL_ARGS="--version 1.0.0 --pad --slot-size 458752"
```

in my `prj.conf`, then build and flash like I normally would

```
west build -b gtsb_ivm21 --pristine; west flash --hex-file build\zephyr\zephyr.signed.confirmed.hex
```
but it still results in hardfault, any idea what I'm missing?