zephyrproject-rtos/zephyr #37502: OPENTHREAD_CUSTOM_PARAMETERS does not seem to work

neye-zhaw, 2021-08-06:
OpenThread defines set with the `OPENTHREAD_CUSTOM_PARAMETERS` Kconfig
option don't seem to take effect.

Steps to reproduce:
1. Set a custom parameter
`CONFIG_OPENTHREAD_CUSTOM_PARAMETERS="OPENTHREAD_CONFIG_MLE_CHILD_TIMEOUT_DEFAULT=610"`
2. Check the value with `otThreadGetChildTimeout()`. The default
   value of 240 is returned.

I'd expect the custom parameters to be applied.

This makes changing some of the OpenThread configuration hard or even impossible.

 - OS: Linux
 - zephyr-sdk-0.12.4
 - zephyr-v2.6.0 (fe7c2efca8)

This affects the https://github.com/zephyrproject-rtos/openthread repo.

rlubos, 2021-08-09:
@neye-zhaw Indeed, OpenThread build system integration, which combines Zephyr and OpenThread build systems, used some obsolete variables from OpenThread build system that were removed at some point. I've submitted a fix in https://github.com/zephyrproject-rtos/zephyr/pull/37532, please have a look.

neye-zhaw, 2021-08-09:
Thanks, the fix works.

PR #37532 (2021-08-10): manifest: Update OpenThread revision
Introduce a bugfix for CONFIG_OPENTHREAD_CUSTOM_PARAMETERS handling in
OpenThread build system integration for Zephyr.

Fixes #37502
closed 2021-08-10 by MaureenHelm