zephyrproject-rtos/zephyr #37520: Is zephyr can run syscall or extrenal program

Rrooach, 2021-08-08:
Hi,

I’m new to zephyr, when I first use it I found that it seems only can flash a prewritten program into the kernel. So I’m wondering if it can directly execute some syscall like a Linux kernel or load an arbitrary binary program into the kernel and execute?

Best,

carlescufi, 2021-08-09:
Hi there. Zephyr currently only supports statically linked applications directly with the kernel.

closed 2021-08-09 by carlescufi

Rrooach, 2021-08-10:
Hi, please let me know if this is possible.
Use qemu to start the kernel. Write an application that can read a specific memory in the system, and this memory can be used as shared memory between qemu and the kernel, which in turn can pass some other programs (like syscall sequences) to the OS ane execute.