zephyrproject-rtos/zephyr #37527: Replace mqtt-azure example with azure-sdk-for-c

xiongyu0523, 2021-08-09:
The current mqtt-azure example project on Zephyr is implemented according to MQTT protocol documented by Microsoft. While Microsoft now provide a very lightweight c sdk (helper) library to handle MQTT protocol stuff and allow user to bring their own connectivity stack (TLS, MQTT, TCP/IP) which should be good fit for Zephyr network, would you mind if I submit a PR to replace the original mqtt-azure project and document with the new one? Thanks.

xiongyu0523, 2021-08-09:
Add Link:
https://github.com/Azure/azure-sdk-for-c

aurel32, 2021-08-09:
MQTT is not a protocol specific to Microsoft, it has been created many years ago by IBM. The protocol being a standard there is no point in using one external implementation per provider.

closed 2021-08-14 by stephanosio