zephyrproject-rtos/zephyr #37536: _pm_devices() skips the very first device in the list and suspend() is not called.

sknaduva, 2021-08-09:
When multiple devices are enabled, suspend() is called for all devices except the very first one -

Suspecting below highlighted line to be checking >= for device_start.

Please acknowledge if this is a genuine issue or I'm missing something?

54 static int _pm_devices(uint32_t state)
55 {
56         const struct device *dev;
57         num_susp = 0;
58
**59         for (dev = (__device_end - 1); dev > __device_start; dev--) {**
60                 bool suspend;
61                 int rc;
62
63                 suspend = should_suspend(dev, state);
64                 if (suspend) {
65                         /*
66                          * Don't bother the device if it is currently
67                          * in the right state.
68                          */
69                         rc = pm_device_state_set(dev, state, NULL, NULL);
70                         if ((rc != -ENOSYS) && (rc != 0)) {
71                                 LOG_DBG("%s did not enter %s state: %d",
72                                         dev->name, pm_device_state_str(state),
73                                         rc);
74                                 return rc;
75                         }
76
77                         __pm_device_slots_start[num_susp] = dev;
78                         num_susp++;
79                 }
80         }
81
82         return 0;
83 }

sknaduva, 2021-08-09:
@gmarull , @ceolin - Does this issue make sense?

gmarull, 2021-08-09:
The code you're listing is not in `main` anymore, can you check latest https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/pm/device.c ?

gmarull, 2021-08-11:
I have checked current implementation in `main` and we iterate over all devices returned by `z_device_get_all_static` correctly. Closing since it doesn't seem like we have a bug, re-open with more details if something else is found.

closed 2021-08-11 by gmarull