zephyrproject-rtos/zephyr #37543: Using STM32Cube HAL function results in linker error

maxgerhardt, 2021-08-09:
**Describe the bug**
I am using PlatformIO to create a Zephyr project for the STM32 Bluepill (STM32F103C8, maps to the [`stm32_mini_dev`](https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v2.6.0/boards/arm/stm32_min_dev) (blue) variant). The base project is https://github.com/platformio/platform-ststm32/tree/develop/examples/zephyr-blink. 

In the firmware I'm trying to use the STM32HAL GPIO functions (such as `HAL_GPIO_Init()`) instead of Zephyr's facilities. So I expand the original code to `#include <stm32f1xx_hal.h>` and call these functions. 

However, building the firmware then fails at the linking stage (`undefined reference to `HAL_GPIO_TogglePin'`).

**To Reproduce**
Steps to reproduce the behavior:
1. Have PlatformIO installed (e.g. [in the CLI](https://docs.platformio.org/en/latest/core/installation.html))
1. Duplicate the [original project](https://github.com/platformio/platform-ststm32/tree/develop/examples/zephyr-blink)
2. Replace `src\main.c` with the code found below
3. Attempt compilation with `pio run`
4. See error

`src\main.c`

```c
#include <zephyr.h>
#include <device.h>
#include <devicetree.h>
#include <drivers/gpio.h>
#include <stm32f1xx_hal.h>

/* 1000 msec = 1 sec */
#define SLEEP_TIME_MS   1000

/* The devicetree node identifier for the "led0" alias. */
#define LED0_NODE DT_ALIAS(led0)

#if DT_NODE_HAS_STATUS(LED0_NODE, okay)
#define LED0    DT_GPIO_LABEL(LED0_NODE, gpios)
#define PIN     DT_GPIO_PIN(LED0_NODE, gpios)
#define FLAGS   DT_GPIO_FLAGS(LED0_NODE, gpios)
#else
/* A build error here means your board isn't set up to blink an LED. */
#error "Unsupported board: led0 devicetree alias is not defined"
#define LED0    ""
#define PIN     0
#define FLAGS   0
#endif

void main(void)
{
        const struct device *dev;
        bool led_is_on = true;
        int ret;

        dev = device_get_binding(LED0);
        if (dev == NULL) {
                return;
        }

        ret = gpio_pin_configure(dev, PIN, GPIO_OUTPUT_ACTIVE | FLAGS);
        if (ret < 0) {
                return;
        }

        __GPIOC_CLK_ENABLE();
        GPIO_InitTypeDef GPIO_InitStruct;
        GPIO_InitStruct.Pin = GPIO_PIN_13;
        GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
        GPIO_InitStruct.Speed = GPIO_SPEED_HIGH;
        HAL_GPIO_Init(GPIOC, &GPIO_InitStruct);

        while (1) {
                gpio_pin_set(dev, PIN, (int)led_is_on);
                led_is_on = !led_is_on;
                HAL_GPIO_TogglePin(GPIOC, GPIO_PIN_13);
                k_msleep(SLEEP_TIME_MS);
        }
}
```

**Expected behavior**
STM32HAL functions are available and compiled in.

**Impact**
Compilation failure, showstopper.

**Logs and console output**
```
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\bluepill_f103c8\src\main.o: in function `main':
C:\Users\Max\temp\zephyr_bluepill/src/main.c:46: undefined reference to `HAL_GPIO_Init'
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.190214/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Max\temp\zephyr_bluepill/src/main.c:51: undefined reference to `HAL_GPIO_TogglePin'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\bluepill_f103c8\zephyr\firmware-pre.elf] Error 1
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK, ...)
 - Commit SHA or Version used: 2.6.0 (stable release)

**Additional context**

When I clean the project I interestingly see some STM32Cube files being compiled, just not `stm32f1xx_hal_gpio.c`..

```
Removed .pio\build\bluepill_f103c8\_pio__modules__hal__stm32__stm32cube\modules\stm32\stm32cube\stm32f1xx\drivers\src\stm32f1xx_hal.c.o
Removed .pio\build\bluepill_f103c8\_pio__modules__hal__stm32__stm32cube\modules\stm32\stm32cube\stm32f1xx\drivers\src\stm32f1xx_hal_rcc.c.o
Removed .pio\build\bluepill_f103c8\_pio__modules__hal__stm32__stm32cube\modules\stm32\stm32cube\stm32f1xx\drivers\src\stm32f1xx_hal_rcc_ex.c.o
Removed .pio\build\bluepill_f103c8\_pio__modules__hal__stm32__stm32cube\modules\stm32\stm32cube\stm32f1xx\drivers\src\stm32f1xx_ll_utils.c.o
Removed .pio\build\bluepill_f103c8\_pio__modules__hal__stm32__stm32cube\modules\stm32\stm32cube\stm32f1xx\soc\system_stm32f1xx.c.o
```

Attempting to set [`CONFIG_USE_STM32_HAL_GPIO=y`](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_USE_STM32_HAL_GPIO.html) in the `zephyr\prj.conf` results in a

```
error: USE_STM32_HAL_GPIO (defined at modules/Kconfig.stm32:189) is assigned in a configuration
file, but is not directly user-configurable (has no prompt).
``` 

error. Adding it in the [board defconfig file](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.6.0/boards/arm/stm32_min_dev/stm32_min_dev_blue_defconfig) results in the same message.

FRASTM, 2021-08-10:
That's right, zephyr project is based on the stm32 LL drivers, not HAL from the stm32Cube. Mainly because the init and configuration steps are done by the zephyr itself with unitary (stm32 LL) functions. 
The stm32 LL is more optimized for the Zephyr than the application-oriented stm32HAL.

maxgerhardt, 2021-08-10:
The configuration file https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/Kconfig.stm32 clearly contains both HAL and LL enabling directives with distinct macros like `USE_STM32_HAL_GPIO` and `USE_STM32_LL_GPIO`. So both HAL and LL code are in there and I can see these source files clearly in https://github.com/zephyrproject-rtos/hal_stm32/tree/main/stm32cube/stm32f1xx/drivers/src and these macros being used in https://github.com/zephyrproject-rtos/hal_stm32/blob/main/stm32cube/stm32f1xx/CMakeLists.txt. 

But somehow including these files is not working.

FRASTM, 2021-08-10:
by a quick and dirty patch, If you force the compilation of the _stm32f1xx_hal_gpio.c_ by changing the 
   `zephyr_library_sources(drivers/src/stm32f1xx_hal_gpio.c)  ` 
in the ../modules/hal/stm32/stm32cube/stm32f1xx
The gpio_hal functions are compiled and also declared in the ../modules/hal/stm32/stm32cube/common_ll/include/stm32_ll_gpio.h by:
```
...
#elif defined(CONFIG_SOC_SERIES_STM32F1X)
#include <stm32f1xx_hal.h>
#include <stm32f1xx_hal_gpio.h>
#include <stm32f1xx_ll_gpio.h>
```
Then the HAL_GPIO_Init  is defined in the src/main.c with a simple 
 #include <stm32_ll_gpio.h>

maxgerhardt, 2021-08-10:
>If you force the compilation of the stm32f1xx_hal_gpio.c by changing the
zephyr_library_sources(drivers/src/stm32f1xx_hal_gpio.c) 

Does this not apply that there's some bug in the build system if the original expression

```
zephyr_library_sources_ifdef(CONFIG_USE_STM32_HAL_GPIO drivers/src/stm32f1xx_hal_gpio.c)
```

is not working, although the config macro should be defined? 

https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/modules/Kconfig.stm32#L9-L14

https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/modules/Kconfig.stm32#L189-L192

FRASTM, 2021-08-11:
I think the `CONFIG_USE_STM32_HAL_GPIO` is missing in the _soc/arm/st_stm32/stm32f1/Kconfig.series_ : 
`	select USE_STM32_HAL_GPIO`

erwango, 2021-08-23:
@maxgerhardt Did you had a look to https://github.com/zephyrproject-rtos/hal_stm32#use-stm32cube-in-your-application ?

maxgerhardt, 2021-08-29:
Indeed, solved in PlatformIO projects by creating the file `zephyr\Kconfig` with content

```
mainmenu "MYAPP"

source "Kconfig.zephyr"
config MYAPP_STM32
  default y
  bool
  select USE_STM32_HAL_GPIO
  select USE_STM32_LL_GPIO
```

closed 2021-08-29 by maxgerhardt