zephyrproject-rtos/zephyr #37571: Bluetooth: Extended advertising assertion

JordanYates, 2021-08-10:
**Describe the bug**

Attempting to use extended advertising in num_events mode (advertise N times then stop) triggers an assertion inside `ull.c` very quickly. Using the timeout mode, also triggers the assertion, but less frequently.
The assertion does not trigger when `BT_LE_ADV_OPT_EXT_ADV` is not set.
This appears to be related to #35013.

**To Reproduce**

Build and run the following sample application, based on the `periodic_adv` sample:
https://github.com/csiro-wsn/zephyr/commit/c7ff12b6b3c6dd6ea5b5df297a202ddf042d8300

**Expected behavior**

No assertions

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.6.0-1913-g497cb2e58795  ***
Starting Extended Advertising Demo
[00:00:00.Next advertising set...
264,495] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.264,526] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.264,526] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.6 Build 99
[00:00:00.265,106] <inf> bt_hci_core: Identity: CA:BC:7E:07:83:13 (random)
[00:00:00.265,106] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:00.265,106] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
Next advertising set...
Next advertising set...
Next advertising set...
Next advertising set...
Next advertising set...
Next advertising set...
Next advertising set...
ASSERTION FAIL [ull_ref_get(ull_hdr)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull.c:2475
[00:00:08.297,271] <err> os: r0/a1:  0x00000003  r1/a2:  0x00010654  r2/a3:  0x00000001
[00:00:08.297,302] <err> os: r3/a4:  0x000026e5 r12/ip:  0x00000000 r14/lr:  0x000066bf
[00:00:08.297,302] <err> os:  xpsr:  0x4100001b
[00:00:08.297,302] <err> os: Faulting instruction address (r15/pc): 0x000066ca
[00:00:08.297,302] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:08.297,302] <err> os: Fault during interrupt handling
[00:00:08.297,302] <err> os: Current thread: 0x20000700 (unknown)
[00:00:08.348,114] <err> os: Halting system
```

**Environment (please complete the following information):**
 - Both the v2.6 release and the latest main branch

cvinayak, 2021-08-10:
This a known (related to uninitialized variables) issue. Fixed in my working branch, I will see if I can cherry-pick it out and send a PR. Please confirm the below commit if cherry-picked into your working copy fixes the issue?

https://github.com/zephyrproject-rtos/zephyr/pull/37387/commits/6c02fd5480198c6f2c872d1278e3c39f56b65a6d

JordanYates, 2021-08-10:
Unfortunately I still get assertions in the same location after applying that commit.

JordanYates, 2021-08-11:
The issue does disappear when enabling `CONFIG_BT_CTLR_LOW_LAT`, but I'm not sure what that means or what the consequences are.

cvinayak, 2021-08-11:
> 
> 
> Unfortunately I still get assertions in the same location after applying that commit.

I am able to reproduce the issue using your sample. It appears the primary and auxiliary PDU scheduling are not synchronously stopped, I will try having a fix today.

`CONFIG_BT_CTLR_LOW_LAT` is an alternative scheduling used with slow CPUs (like nRF51), with a consequence of reduced radio utilization. This option may be hiding the issue.

PR #37594 (2021-08-20): Bluetooth: Controller: Fix sync of primary and auxiliary adv terminate
Fix missing synchronization when stopping primary and
auxiliary PDU scheduling.

Added implementation to ensure primary and auxiliary event
count match when terminating extended advertising on maximum
number of events or on duration.

Fixes #37571.
commit 80f382e5 (2021-08-20): Bluetooth: Controller: Fix sync of primary and auxiliary adv terminate

Fix missing synchronization when stopping primary and
auxiliary PDU scheduling.

Added implementation to ensure primary and auxiliary event
count match when terminating extended advertising on maximum
number of events or on duration.

Fixes #37571.
closed 2021-08-20 by carlescufi