zephyrproject-rtos/zephyr #37581: Bluetooth: controller: radio: Change CTE configuration method

saleh-unikie, 2021-08-10:
**Is your enhancement proposal related to a problem? Please describe.**
In direction finding applications, current source code uses direct access to radio registers to set CTE related configuration.
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio_df.c#L256

while it is possible to use functions provided by the HAL. It is more abstract and the final result increases the readablity.

**Describe the solution you'd like**
I have provided a solution here:
https://github.com/zephyrproject-rtos/zephyr/pull/37504

**Describe alternatives you've considered**
N/A

**Additional context**
A similar approach has been used in `radio_df_cte_inline_set_enabled` function:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio_df.c#L223

ppryga, 2021-08-12:
Radio related functions are used in interrupt context. Policy for that code (at least in Nordic lower link layer implementation) is to go for the lowest possible number of executed instructions. 
I didn't use HAL `nrf_radio_dfectrl_configure` function to configure `DFECTRL1` register because the function updates also `DFECTRL2`. 
`DFECTRL2` register is responsible for fine-tuning of switch and sample offsets. This is a feature provided in case someone uses slow RF switches. 
Current implementation does not update the `DFECTRL2` register because default values were good enough. 
Implementation of switch and sample offset in is not in my current scope. If you need this fine tuning configuration, then you are welcome to provide a PR with that.

If the `DFECTRL2` is not used by the implementation then we save some instructions by direct access to `DFECTRL1`. This sacrifices a bit readability in favour of performance.

saleh-unikie, 2021-08-12:
Thanks for your reply :)

As you know, codes are more read, rather than to be written. so maybe if the performance change is negligible, increasing the readability can help the maintenance of the project.
Another issue is "abstraction", which also again affect on the maintenance cost.

Anyway, the default value of `DFECTRL2` is 0 as you can find in the below link, and this value has been used to initialize this register:
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf52811%2Fradio.html&cp=4_4_0_5_13_14_79&anchor=register.DFECTRL2

My suggestion is to merge this changes, and then it is ready to develop a "configuration option" PR to choose "fine-tuning" values by user or leave it as default.

saleh-unikie, 2021-08-12:
PS: In addition to the provided changes, a minor bug has fixed in the above commit, third parameter of `radio_df_ctrl_set` function should be "TSAMPLESPACING" while "TSAMPLESPACINGREF" has been passed.

ppryga, 2021-08-12:
>As you know, codes are more read, rather than to be written. so maybe if the performance change is negligible, increasing the readability can help the maintenance of the project.
Another issue is "abstraction", which also again affect on the maintenance cost.

I know that. Due to that and due to our policy I've verified cost of each function. As you have already spotted, I've been using HAL  function calls in other places. In this case write to `DFECTRL2` are not necessary, this is additional cost of the call.

As I've written that is a policy to go for performance scarifying (a bit) readability. 

> PS: In addition to the provided changes, a minor bug has fixed in the above commit, third parameter of radio_df_ctrl_set function should be "TSAMPLESPACING" while "TSAMPLESPACINGREF" has been passed.

Update it please in former version of the code in the PR. BTW. Fortunately the values are the same for both #defines.

saleh-unikie, 2021-08-16:
A fix has provided for the minor issue that was mentioned (Replacing TSAMPLESPACINGREF by TSAMPLESPACING):
https://github.com/zephyrproject-rtos/zephyr/pull/37715

closed 2021-08-17 by carlescufi