zephyrproject-rtos/zephyr #37586: get_maintainer.py is broken

mbolivar-nordic, 2021-08-10:
**Describe the bug**

This can't be used to get the maintainer for a path.

**To Reproduce**

`$ ./scripts/get_maintainer.py path .`

This reports an error about some random line in the file. If you fix the error and run it again, you will get another error. This keeps happening for many lines in the file, so either we've just never used this script and it doesn't actually work or the file is malformed.

mbolivar-nordic, 2021-08-10:
@nashif you're listed in CODEOWNERS for get_maintainer.py; am I just holding it wrong?

mbolivar-nordic, 2021-08-10:
Concrete example: 

```
$ ./scripts/get_maintainer.py path drivers/i2c/i2c_tca9546a.c 
./scripts/get_maintainer.py: error: /home/mbolivar/zp/zephyr/MAINTAINERS.yml: glob pattern 'arch/arm/aarch32/' in 'files' in area 'ARM arch' does not match any files
```

aescolar, 2021-08-11:
Duplicate of #32531 :)

closed 2021-08-11 by carlescufi