zephyrproject-rtos/zephyr #37597: samples: bluetooth: scan_adv

ycsin, 2021-08-11:
**Describe the bug**
I'm not sure if I'm missing something, but when I compile and flash the `scan_adv` sample to the `nrf52833dk_nrf52833` board, I get nothing on the terminal.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52833dk_nrf52833 samples/bluetooth/scan_adv
2. west flash
3. Check terminal

**Expected behavior**
I expect the terminal to show something, and then it should appear in my phone's scanned BLE list, just like the `periodic_adv` sample.

**Impact**
Unable to test `nrf52833dk_nrf52833`

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain gnuarmembi
 - v2.6.0-1638-g0296d12675fd

carlescufi, 2021-08-12:
@ppryga could you try running this sample in the nrf52833dk_nrf52833  DK please?

ppryga, 2021-08-13:
I've verified the issue.

The sample builds and runs on the `nrf52833dk_nrf52833`.
```
*** Booting Zephyr OS build zephyr-v2.6.0  ***
Starting Scanner/Advertiser Demo
Bluetooth initialized
[00:00:00.389,404] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.389,404] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.389,404] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.6 Build 0
[00:00:00.389,953] <inf> bt_hci_core: Identity: E3:C4:81:0B:B5:E6 (random)
[00:00:00.389,953] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:00.389,953] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
```

There is another thing. I've got an assert:
```

ASSERTION FAIL [evdone] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:353
[00:20:46.116,149] <err> os: r0/a1:  0x00000003  r1/a2:  0x0000fc08  r2/a3:  0x00000001
[00:20:46.116,180] <err> os: r3/a4:  0x00001e81 r12/ip:  0x00000000 r14/lr:  0x000083bf
[00:20:46.116,180] <err> os:  xpsr:  0x41000011
[00:20:46.116,180] <err> os: Faulting instruction address (r15/pc): 0x000083ca
[00:20:46.116,180] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:20:46.116,180] <err> os: Fault during interrupt handling

[00:20:46.116,180] <err> os: Current thread: 0x20000770 (unknown)
[00:20:46.829,193] <err> os: Halting system
```
According to @cvinayak that is a solved issue. 
I'll run the sample with lates main branch to double check that it is working correctly.

ppryga, 2021-08-13:
Update! @cvinayak the issue with assert is reproducible on current main also: 7b9fbcd3b7

cvinayak, 2021-08-13:
@ppryga could you paste the assert log with line number and the Faulting instruction address here so that I can try reproduce the same build.

ppryga, 2021-08-13:
Here is log:
```
ASSERTION FAIL [evdone] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:352
[00:27:12.492,126] <err> os: r0/a1:  0x00000003  r1/a2:  0x00010038  r2/a3:  0x00000001
[00:27:12.492,156] <err> os: r3/a4:  0x00001f0d r12/ip:  0x00000000 r14/lr:  0x0000869f
[00:27:12.492,156] <err> os:  xpsr:  0x41000011
[00:27:12.492,156] <err> os: Faulting instruction address (r15/pc): 0x000086aa
[00:27:12.492,156] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:27:12.492,156] <err> os: Fault during interrupt handling

[00:27:12.492,156] <err> os: Current thread: 0x200006f8 (unknown)
[00:27:13.205,291] <err> os: Halting system
```

Here is actual code line that fails:

PR #37700 (2021-08-16): Bluetooth: Controller: Only remove duplicate resume events
When events are placed back into pipeline as resume events,
only remove duplicate resume events and not new prepare
events that may have been enqueue between the start of the
pre-empt ticker and its timeout.

Due to new prepare events that was removed, extra done
events generated cause the number of enqueued done events
to overflow and assert.

Fixes #36381.
Fixes #37597.

Controller debug pin toggle detecting a late advertising event caused by a race condition where new scan prepare is enqueued between adv prepare and its pre-empt tim [... cut]
commit 190532bc (2021-08-16): Bluetooth: Controller: Only remove duplicate resume events

When events are placed back into pipeline as resume events,
only remove duplicate resume events and not new prepare
events that may have been enqueue between the start of the
pre-empt ticker and its timeout.

Due to new prepare events that was removed, extra done
events generated cause the number of enqueued done events
to overflow and assert.

Fixes #36381.
Fixes #37597.
closed 2021-08-16 by cfriedt