zephyrproject-rtos/zephyr #37600: Invalidate TLB after ptables swap

manuargue, 2021-08-11:
I'm running Zephyr on a custom SoC which has a cluster of 2 Cortex-A53 and a cache-coherence interconnect. I found the following issue whe executing the test _tets/kernel/mem_protect/syscalls_, test case _test_mem_domain_migration_.

After the child thread is migrated to the new domain and k_thread_join is invoked, once the child thread is switched it still "sees" old page table entries which were cached in the TLB, resulting in a fatal error (data abort). This seems to happen because z_arm64_swap_ptables is not invalidating the TLB after switching to the new the page tables, which may share entries with the old table just swapped. Adding TLB invalidation on z_arm64_swap_ptables after setting new value for TTBR0 fixes the issue.

I couldn't reproduce this issue with qemu_cortex_a53, that's why I'm not logging this as a bug. But my understanding is that the TLB should be invalidated after updating TTBR0 to avoid this kind of situation. In a similar way that TLB are being invalidated after privatizing or resetting maps, whenever the page table is active.

cfriedt, 2021-08-17:
@carlocaione - can you please assign a a priority to this issue?

carlocaione, 2021-08-23:
This was fixed with #37601

closed 2021-08-23 by carlocaione