zephyrproject-rtos/zephyr #37602: I2C drivers use inconsistent init priorities

Flauer741, 2021-08-11:
**Describe the bug**
The majority of i2c drivers use a init-priority of CONFIG_I2C_INIT_PRIORITY (default 40), but some drivers e.g. i2c_ll_stm32 use CONFIG_KERNEL_INIT_PRIORITY_DEVICE and ignore CONFIG_I2C_INIT_PRIORITY. This leads to unexpected behavior for users setting I2C_INIT_PRIORITY and expecting it to be applied. The other issue is inconsistent behavior of software across different boards, especially if there are i2c dependent drivers with a priority between 40 and 50.

**Expected behavior**
All I2C drivers use the same init priority, preferable CONFIG_I2C_INIT_PRIORITY. Alternatively CONFIG_I2C_INIT_PRIORITY might be removed and all drivers be converted to CONFIG_KERNEL_INIT_PRIORITY_DEVICE.
As a minimum I would expect a documentation for the CONFIG_I2C_INIT_PRIORITY of when/why this option is/is-not used.

**Impact**
annoyance. Manually setting priorities in prj.conf required.

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK 0.12.4
 - Zephyr 2.6.1-rc1

gmarull, 2021-08-11:
Seems like we have a wide variety of combinations in I2C...
```
drivers/i2c/i2c_npcx_port.c:                        PRE_KERNEL_1, CONFIG_KERNEL_INIT_PRIORITY_DEVICE,  \
drivers/i2c/i2c_mcux.c:                 CONFIG_KERNEL_INIT_PRIORITY_DEVICE,             \
drivers/i2c/i2c_gecko.c:                 POST_KERNEL, CONFIG_KERNEL_INIT_PRIORITY_DEVICE, \
drivers/i2c/i2c_mchp_xec.c:             POST_KERNEL, CONFIG_I2C_INIT_PRIORITY,                  \
drivers/i2c/i2c_rv32m1_lpi2c.c:                     POST_KERNEL, CONFIG_I2C_INIT_PRIORITY,             \
drivers/i2c/Kconfig.tca9546a:   default I2C_INIT_PRIORITY
drivers/i2c/Kconfig.tca9546a:   default I2C_INIT_PRIORITY
drivers/i2c/i2c_sbcon.c:            PRE_KERNEL_2, CONFIG_I2C_INIT_PRIORITY, &api);
drivers/i2c/i2c_sam_twihs.c:                        POST_KERNEL, CONFIG_I2C_INIT_PRIORITY,      \
drivers/i2c/i2c_sam0.c:                     CONFIG_I2C_INIT_PRIORITY,                   \
drivers/i2c/i2c_ll_stm32.c:                 POST_KERNEL, CONFIG_KERNEL_INIT_PRIORITY_DEVICE,    \
drivers/i2c/i2c_lpc11u6x.c:                 PRE_KERNEL_1, CONFIG_KERNEL_INIT_PRIORITY_OBJECTS,        \
drivers/i2c/i2c_emul.c:                     CONFIG_I2C_INIT_PRIORITY, \
drivers/i2c/i2c_mcux_flexcomm.c:                            CONFIG_KERNEL_INIT_PRIORITY_DEVICE,         \
drivers/i2c/i2c_mcux_lpi2c.c:                       CONFIG_KERNEL_INIT_PRIORITY_DEVICE,         \
drivers/i2c/i2c_gpio.c:     POST_KERNEL, CONFIG_I2C_INIT_PRIORITY, &api);
drivers/i2c/i2c_rcar.c:                       POST_KERNEL, CONFIG_KERNEL_INIT_PRIORITY_DEVICE, \
drivers/i2c/i2c_sifive.c:                           CONFIG_I2C_INIT_PRIORITY, \
drivers/i2c/i2c_cc13xx_cc26xx.c:                POST_KERNEL, CONFIG_I2C_INIT_PRIORITY,
drivers/i2c/i2c_sam_twi.c:                          POST_KERNEL, CONFIG_I2C_INIT_PRIORITY,      \
drivers/i2c/slave/eeprom_slave.c:                           CONFIG_I2C_SLAVE_INIT_PRIORITY,             \
drivers/i2c/i2c_nios2.c:                    POST_KERNEL, CONFIG_KERNEL_INIT_PRIORITY_DEVICE,
drivers/i2c/i2c_test.c:                       CONFIG_KERNEL_INIT_PRIORITY_DEVICE,       \
drivers/i2c/i2c_litex.c:                           CONFIG_I2C_INIT_PRIORITY,                           \
drivers/i2c/i2c_dw.c:                         POST_KERNEL, CONFIG_I2C_INIT_PRIORITY,          \
drivers/i2c/i2c_ite_it8xxx2.c:                  CONFIG_KERNEL_INIT_PRIORITY_DEVICE,                    \
drivers/i2c/i2c_esp32.c:                    POST_KERNEL, CONFIG_I2C_INIT_PRIORITY,
drivers/i2c/i2c_esp32.c:                    POST_KERNEL, CONFIG_I2C_INIT_PRIORITY,
drivers/i2c/i2c_npcx_controller.c:                          PRE_KERNEL_1, CONFIG_KERNEL_INIT_PRIORITY_DEVICE,  \
drivers/i2c/i2c_sam4l_twim.c:                       POST_KERNEL, CONFIG_I2C_INIT_PRIORITY,      \
drivers/i2c/i2c_cc32xx.c:                   POST_KERNEL, CONFIG_KERNEL_INIT_PRIORITY_DEVICE,
drivers/i2c/i2c_nrfx_twim.c:                  CONFIG_I2C_INIT_PRIORITY,                                \
drivers/i2c/i2c_imx.c:                          CONFIG_KERNEL_INIT_PRIORITY_DEVICE,     \
drivers/i2c/i2c_nrfx_twi.c:                   CONFIG_I2C_INIT_PRIORITY,                                \

```

I assume all should be moved to `POST_KERNEL` + `CONFIG_I2C_INIT_PRIORITY`

nashif, 2021-09-29:
I think this is a bug, the driver should not be using CONFIG_KERNEL_INIT_PRIORITY_DEVICE

closed 2021-12-15 by github-actions[bot]