zephyrproject-rtos/zephyr #37608: Stack overflow when logging float

kurddt, 2021-08-11:
**Describe the bug**
When using log to print a float variable, even if trans-typed to an integer, a stack overflow is detected. When the type of the variable id changed to int no stack overflow is detected. The same code run without issue using zephyr v2.4.0, but crash with v2.5.0

**To Reproduce**
Source code:
```
#include <logging/log.h>
#include <logging/log_ctrl.h>

LOG_MODULE_REGISTER(MAIN, CONFIG_MAIN_LOG_LEVEL);

#if 1
float test = 0;
#else
int test = 0;
#endif
_Noreturn void system_monitor()
{
	LOG_DBG("Started \n");

	while(true) {
		k_sleep(K_SECONDS(1));
		LOG_INF("New sec state %u", 0);

		size_t unused_stack;
		k_thread_stack_space_get(k_current_get(), &unused_stack);
		LOG_INF("Stack words remaining: %d", unused_stack);

		int angle = test;
		LOG_INF("Inclination: %u", angle);

		k_thread_stack_space_get(k_current_get(), &unused_stack);
		LOG_INF("Stack words remaining: %d", unused_stack);
	}
}

int main(void)
{
	log_init();

	system_monitor();
}
```

prj.conf:
```
CONFIG_STDOUT_CONSOLE=n
CONFIG_PS2=n
CONFIG_LOG=y
CONFIG_LOG_IMMEDIATE_CLEAN_OUTPUT=n
CONFIG_LOG_BACKEND_RTT=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_NULL_POINTER_EXCEPTION_DETECTION_MPU=y

CONFIG_FPU=y
CONFIG_FPU_SHARING=y
CONFIG_THREAD_STACK_INFO=y
CONFIG_STACK_SENTINEL=y
CONFIG_STACK_USAGE=y
CONFIG_INIT_STACKS=y
CONFIG_LOG_DEFAULT_LEVEL=4
#CONFIG_MAIN_STACK_SIZE=10000
#CONFIG_ISR_STACK_SIZE=10000

CONFIG_NFCT_PINS_AS_GPIOS=y

CONFIG_NUM_COOP_PRIORITIES=1
CONFIG_NUM_PREEMPT_PRIORITIES=4
CONFIG_MAIN_THREAD_PRIORITY=3
```

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.13.1)

# Set some global paths and variable
set(BOARD nrf52dk_nrf52832)
set(ZEPHYR_CMAKE ${CMAKE_SOURCE_DIR}/../zephyr/share/zephyr-package/cmake/)

message(status "Board ROOT: ${BOARD_ROOT}")

list(APPEND CMAKE_PREFIX_PATH "${ZEPHYR_CMAKE}") #With this we don't need to install inside the user registry
# Get required CMake packages
find_package(Zephyr REQUIRED)

# All previous statements need to stay before project definition (for some reason)
project(Mems_demo)

target_sources(app PRIVATE
	app/main.c
)

target_compile_definitions(app PRIVATE
	CONFIG_MAIN_LOG_LEVEL=4)

target_compile_options(app PRIVATE -Wextra -Wno-unused-parameter -Werror)

```

Steps to reproduce the behavior:
Using the attached project files
1. mkdir build; cd build
2. cmake ../
3. make
4. Run on target
5. See error

[CMakeLists.txt](https://github.com/zephyrproject-rtos/zephyr/files/6967628/CMakeLists.txt)
**Logs and console output**
```
[00:00:01.268,249] <inf> MAIN: New sec state 0
[00:00:01.268,432] <inf> MAIN: Stack words remaining: 740
[00:00:01.268,432] <inf> MAIN: Inclination: 0
[00:00:01.268,554] <inf> MAIN: Stack words remaining: 740
[00:00:02.290,557] <inf> MAIN: New sec state 0
[00:00:02.290,710] <inf> MAIN: Stack words remaining: 644
[00:00:02.290,710] <inf> MAIN: Inclination: 0
[00:00:02.290,832] <inf> MAIN: Stack words remaining: 644
[00:00:02.290,863] <err> os: r0/a1:  0x00000002  r1/a2:  0x00000000  r2/a3:  0xf0f0f0f0
[00:00:02.290,863] <err> os: r3/a4:  0x20000e00 r12/ip:  0x20000050 r14/lr:  0x00006695
[00:00:02.290,863] <err> os:  xpsr:  0x41000000
[00:00:02.290,863] <err> os: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
[00:00:02.290,893] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
[00:00:02.290,893] <err> os: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
[00:00:02.290,924] <err> os: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
[00:00:02.290,924] <err> os: fpscr:  0x00000000
[00:00:02.290,924] <err> os: Faulting instruction address (r15/pc): 0x00004946
[00:00:02.290,924] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:02.290,954] <err> os: Current thread: 0x20000208 (unknown)
[00:00:02.856,933] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: GNU Tools for Arm Embedded Processors 7-2017-q4-major
 - Commit SHA:  da8f9d77c4

kurddt, 2021-08-11:
I'm wondering if this issue is not triggered by a stack sentinel not being written at the correct offset. When debugging I see the following:
```
	stack = (uint32_t *)_current->stack_info.start;
	if (*stack != STACK_SENTINEL) {
		/* Restore it so further checks don't trigger this same error */
		*stack = STACK_SENTINEL;
		z_except_reason(K_ERR_STACK_CHK_FAIL);
	}
```

```
(gdb) p /x _kernel.cpus[0].current->stack_info.start
$12 = 0x20000900
```

```
0x20000880 <z_main_stack>:	0xcb1ed56c	0x06b52f40	0xecba1ba4	0xea9b36bd
0x20000890 <z_main_stack+16>:	0x28d0d9bf	0x3144de61	0xbd3c507b	0x1df21885
0x200008a0 <z_main_stack+32>:	0xf0f0f0f0	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa
0x200008b0 <z_main_stack+48>:	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa
0x200008c0 <z_main_stack+64>:	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa
0x200008d0 <z_main_stack+80>:	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa
0x200008e0 <z_main_stack+96>:	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa
0x200008f0 <z_main_stack+112>:	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa
0x20000900 <z_main_stack+128>:	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa	0xaaaaaaaa
```

So it seems that the sentinel is written to `0x0x200008a0` but `stack_info.star == 0x20000900`

kurddt, 2021-08-11:
Faulty code is here:
https://github.com/zephyrproject-rtos/zephyr/blob/0f949caee12441dd4996017210d2d05971b3cf23/arch/arm/core/aarch32/thread.c#L159

This change the stack start inside the stack info structure of the thread making it incompatible with `STACK_SENTINEL`

carlescufi, 2021-08-11:
@kurddt perhaps you could consider sending a PR with a fix so that the relevant arch owner can take a look?

kurddt, 2021-08-12:
I'm not quite sure how to fix that yet.

kurddt, 2021-08-12:
Maybe @ioannisg have some idea how to fix that since he last modified those lines: https://github.com/zephyrproject-rtos/zephyr/commit/cafe04558cedf583572b60d0a2b1b0161c89ac5f

kurddt, 2021-08-19:
One solution would be to read the value pointed by thread->stack_info.start before it's modified and write it back afterward. But I'm not sure it's the smartest thing to do

stephanosio, 2021-09-07:
@nordic-krch Have you had a chance to look into this?

nordic-krch, 2021-09-07:
No, didn't yet look into it. Will take a look in upcoming days.

nordic-krch, 2021-09-08:
Looking at it a bit, as mentioned above, it does not look like a logging issue. Issue occurs after enabling stack sentinel.

@ioannisg any ideas?

cfriedt, 2021-09-23:
I had to increase shell stack depth beyond the default recently for a simple `LOG_INF()` in a loop from samples/hello_world. Was 2048, I increased to 4096. It was a 64-bit target. originally, I thought it was a logging issue but it was shell.

Just thought it might be related

kurddt, 2021-09-23:
The issue is when using both stack sentinel + MPU stack protectio. I will change the title of the issue

closed 2021-12-07 by github-actions[bot]