zephyrproject-rtos/zephyr #37610: subsys/mgmt/hawkbit: Unable to parse json if the payload is split into 2 packets

ycsin, 2021-08-11:
**Describe the bug**
`response_cb` isn't have to store the full packet from `HAWKBIT_PROBE` & `HAWKBIT_PROBE_DEPLOYMENT_BASE` properly if they were to arrive in more than a single packet. I'm guessing that this line is causing the issue, but I don't really understand it:

```c
if ((body_data == NULL) && (final_data == HTTP_DATA_MORE)) {
```

However I understand the [`HAWKBIT_DOWNLOAD` part](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L783) and know this it should be applicable to the `HAWKBIT_PROBE` & `HAWKBIT_PROBE_DEPLOYMENT_BASE`. After applying this part solves the issue. I believe we can can use the same code to extract the body for all cases:

```c
if (hb_context.dl.http_content_size == 0) {
	body_data = rsp->body_start;
	body_len = rsp->data_len;
	/*
	 * subtract the size of the HTTP header from body_len
	 */
	body_len -= (rsp->body_start - rsp->recv_buf);
	hb_context.dl.http_content_size = rsp->content_length;
} else {
	/*
	 * more general case where body data is set, but no need
	 * to take the HTTP header into account
	 */
	body_data = rsp->body_start;
	body_len = rsp->data_len;
}

if ((rsp->body_found == 1) && (body_data == NULL)) {
	body_data = rsp->recv_buf;
	body_len = rsp->data_len;
}
```

ycsin, 2021-08-11:
@Navin-Sankar @burumaj Any comments?

Navin-Sankar, 2021-08-12:
@ycsin yes `HAWKBIT_DOWNLOAD` part code is applicable to both the `HAWKBIT_PROBE` and `HAWKBIT_PROBE_DEPLOYMENT_BASE`. 
Or try to collect the response data in single packet by changing the [RECV_BUFFER_SIZE](https://github.com/zephyrproject-rtos/zephyr/blob/a42890fb394d197bfac4c48052241c8aee3d8c04/subsys/mgmt/hawkbit/hawkbit.c#L43)

ycsin, 2021-08-12:
@Navin-Sankar I think we should fix the root cause and unify how the `response_cb` extracts `body_data` and `body_len`, what do you think?

@burumaj Would it be possible for you to submit another PR on this? Maybe a have a `body_extract` function that has your latest changes in it which can be used by all three cases?

burumaj, 2021-08-12:
>I just tried to increase RECV_BUFFER_SIZE to 1024, but still receives the payload in chunks. I'm not very familiar over how http works, but this chunked transfer seemed to be controlled by the host and is a feature of HTTP 1.1.

@ycsin If i recall correctly, this has to do with the network code. What happens is there is some remaining data in a network internal buffer that needs to be consumed before another "frame" of data can be retrieved. This is what I think is causing the issue you're seeing. This behavior is because of how the network driver is implemented. If you're using a modem I recommend looking into the implementation of the ofloaded network calls like `read()`

ycsin, 2021-08-12:
@burumaj But I'm using a `gsm_ppp` modem, which shouldn't be using offloaded stuffs.

>What happens is there is some remaining data in a network internal buffer that needs to be consumed before another "frame" of data can be retrieved.

I'm also finding where this happens, I guess this could be the reason why the download is slow.
@jukkar May I have your inputs please?

ycsin, 2021-08-17:
@burumaj Do you mind submitting another PR that basically copy your latest implementation and apply it to download and deploymentbase parsing case?

burumaj, 2021-08-18:
I don't mind submitting another pr, but I can't promise anything timeline related.

closed 2021-09-14 by cfriedt