zephyrproject-rtos/zephyr #37611: Implement L2CAP ecred reconfiguaration request as initiator

KKopyscinski, 2021-08-11:
**Is your enhancement proposal related to a problem? Please describe.**
Currently reconfiguaration request can only be received and not send. The problem is that this should be supported, as Zephyr declares `Enhanced Credit Based Flow Control Channel Initiator` feature to be supported (`TSPC_L2CAP_1_5	LE Data Channel Initiator` and `TSPC_L2CAP_2_48 Support Enhanced Credit Based Flow Control Mode` are supported). Lack of implemented functions and API don't allow us to pass L2CAP/ECFC/BV-22-C.

**Describe the solution you'd like**
Mechanism of initiating L2CAP Enhanced channel reconfiguration should be implemented (send request and process response)

**Describe alternatives you've considered**
I don't see any alternatives when support of functionality is declared.

**Additional context**
Red rectangle marks what should be implemented according to Test Specifiacation
[image]

sjanc, 2021-09-15:
Since this seems to be mostly for qualification purposes (at least until now no user reported need for being able to reconfigure channel) we can try make is simple and hide it under CONFIG_BT_TESTING.

So my proposal would be to add CONFIG_BT_L2CAP_ECRED_LAZY_RECONF (or whatever name fits) that would cause channels to be created with default MPS/MTU and have new test api int bt_l2cap_chan_reconfigure(struct bt_l2cap_chan *chan); that would allow to reconfigure tx MTU/MPS to whatever is set by tx endpoint.

sjanc, 2021-09-17:
L2CAP/ECFC/BV-22-C is fixed by https://github.com/zephyrproject-rtos/zephyr/pull/38639

L2CAP/ECFC/BV-24-C can be waived as IUT doesn't support on-demand MPS decreasing

PR #38639 (2021-10-04): Bluetooth: L2CAP channels reconfiguration support
Fixes #37611
closed 2021-10-04 by carlescufi