zephyrproject-rtos/zephyr #37637: Infinite configuring loop for samples\drivers\led_ws2812 sample

thedjnK, 2021-08-12:
**Describe the bug**
Upon configuring the sample application samples\drivers\led_ws2812 for the nrf52dk_nrf52832 target, it configures OK, when building with ninja however, it endlessly re-configures the project over and over

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52dk_nrf52832 ..
3. ninja
4. See loop

**Expected behavior**
The project to build

**Impact**
Impact wise the sample is useless, so showstopped in that regard, but in terms of priority, probably very low

**Logs and console output**
Configuring:
```
cmake -GNinja -DBOARD=nrf52dk_nrf52832 ..
Including boilerplate (Zephyr base): C:/zephy_5/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/zephy_5/zephyr/samples/drivers/led_ws2812
-- Zephyr version: 2.6.0-rc1 (C:/zephy_5/zephyr), build: v2.6.0-rc1-ncs1-6-gf1cb56cfb023
-- Found Python3: C:/Python38/python.exe (found suitable exact version "3.8.10") found components: Interpreter
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- Board: nrf52dk_nrf52832
-- Cache files will be written to: C:/zephy_5/zephyr/.cache
-- Found toolchain: gnuarmemb (C:/gcc-arm)
-- Found BOARD.dts: C:/zephy_5/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832.dts
-- Found devicetree overlay: C:/zephy_5/zephyr/samples/drivers/led_ws2812/boards/nrf52dk_nrf52832.overlay
-- Generated zephyr.dts: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/include/generated/device_extern.h
Parsing C:/zephy_5/zephyr/samples/drivers/led_ws2812/Kconfig
Loaded configuration 'C:/zephy_5/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832_defconfig'
Merged configuration 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/prj.conf'
Merged configuration 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/boards/nrf52dk_nrf52832.conf'
Configuration saved to 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/.config'
Kconfig header saved to 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/gcc-arm/bin/arm-none-eabi-gcc.exe
CMake Warning at ../../../soc/arm/nordic_nrf/nrf52/CMakeLists.txt:21 (message):
  Both SOC_NRF52832_ALLOW_SPIM_DESPITE_PAN_58 and an NRF SPIM driver are
  enabled, therefore PAN 58 will apply if RXD.MAXCNT == 1 and TXD.MAXCNT <= 1

-- Application version: v2.6.0-rc1-ncs1-6-gf1cb56cfb023
-- Configuring done
-- Generating done
-- Build files have been written to: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build
```

Building:
```
ninja
[0/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): C:/zephy_5/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/zephy_5/zephyr/samples/drivers/led_ws2812
-- Zephyr version: 2.6.0-rc1 (C:/zephy_5/zephyr), build: v2.6.0-rc1-ncs1-6-gf1cb56cfb023
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- Board: nrf52dk_nrf52832
-- Cache files will be written to: C:/zephy_5/zephyr/.cache
-- Found toolchain: gnuarmemb (C:/gcc-arm)
-- Found BOARD.dts: C:/zephy_5/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832.dts
-- Found devicetree overlay: C:/zephy_5/zephyr/samples/drivers/led_ws2812/boards/nrf52dk_nrf52832.overlay
-- Generated zephyr.dts: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/include/generated/device_extern.h
Parsing C:/zephy_5/zephyr/samples/drivers/led_ws2812/Kconfig
Loaded configuration 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/.config'
No change to configuration in 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/.config'
No change to Kconfig header in 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/include/generated/autoconf.h'
CMake Warning at ../../../soc/arm/nordic_nrf/nrf52/CMakeLists.txt:21 (message):
  Both SOC_NRF52832_ALLOW_SPIM_DESPITE_PAN_58 and an NRF SPIM driver are
  enabled, therefore PAN 58 will apply if RXD.MAXCNT == 1 and TXD.MAXCNT <= 1

-- Application version: v2.6.0-rc1-ncs1-6-gf1cb56cfb023
-- Configuring done
-- Generating done
-- Build files have been written to: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build
[0/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): C:/zephy_5/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/zephy_5/zephyr/samples/drivers/led_ws2812
-- Zephyr version: 2.6.0-rc1 (C:/zephy_5/zephyr), build: v2.6.0-rc1-ncs1-6-gf1cb56cfb023
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- Board: nrf52dk_nrf52832
-- Cache files will be written to: C:/zephy_5/zephyr/.cache
-- Found toolchain: gnuarmemb (C:/gcc-arm)
-- Found BOARD.dts: C:/zephy_5/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832.dts
-- Found devicetree overlay: C:/zephy_5/zephyr/samples/drivers/led_ws2812/boards/nrf52dk_nrf52832.overlay
-- Generated zephyr.dts: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/include/generated/device_extern.h
Parsing C:/zephy_5/zephyr/samples/drivers/led_ws2812/Kconfig
Loaded configuration 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/.config'
No change to configuration in 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/.config'
No change to Kconfig header in 'C:/zephy_5/zephyr/samples/drivers/led_ws2812/build/zephyr/include/generated/autoconf.h'
CMake Warning at ../../../soc/arm/nordic_nrf/nrf52/CMakeLists.txt:21 (message):
  Both SOC_NRF52832_ALLOW_SPIM_DESPITE_PAN_58 and an NRF SPIM driver are
  enabled, therefore PAN 58 will apply if RXD.MAXCNT == 1 and TXD.MAXCNT <= 1

-- Application version: v2.6.0-rc1-ncs1-6-gf1cb56cfb023
-- Configuring done
ninja: error: rebuilding 'build.ninja': interrupted by user
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: ARM GCC GNU 10.3.1 2021.07
 - Commit SHA or Version used: nRF connect SDK 1.6.1 with the version of zephyr that has

mbolivar-nordic, 2021-09-01:
```
-- Application version: v2.6.0-rc1-ncs1-6-gf1cb56cfb023
```

@thedjnK This is not an upstream Zephyr version. Please retest on upstream Zephyr and reopen if it reproduces.

mbolivar-nordic, 2021-09-01:
@thedjnK if it only reproduces downstream, please use DevZone to let us know.

epietrowicz, 2021-09-03:
I was able to reproduce using upstream (see attached for build logs). What can I do to work around this error? 
`Zephyr version: 2.6.99 (C:/Users/Jatin/zephyrproject/zephyr), build: zephyr-v2.6.0-2940-g529d9149d395`

[console_output_looping_bug.txt](https://github.com/zephyrproject-rtos/zephyr/files/7107688/console_output_looping_bug.txt)

tejlmand, 2021-09-06:
Fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/38333

PR #38333 (2021-09-11): cmake: fixing toolchain_parse_make_rule to correctly handle `\` files
Fixes: #37637

toolchain_parse_make_rule() parses depfiles and converted `\\` to the
CMake list separator `;`.

However, gcc -M might create depfiles with windows path separator `\`
causing this conversion to fail, as a path like:
```
c:\...\zephyr\samples\drivers\led_ws2812\nrf52-bindings.h
```

resulting in a file list as:
```
c:;...;zephyr;samples;drivers;led_ws2812;nrf52-bindings.h
```

which results in a CMake configure dependency to be added to `C:`.
As `C:` is always newer than the build.ninja file, this resulted in
c [... cut]
commit 14b63148 (2021-09-11): cmake: fixing toolchain_parse_make_rule to correctly handle `\` files

Fixes: #37637

toolchain_parse_make_rule() parses depfiles and converted `\\` to the
CMake list separator `;`.

However, gcc -M might create depfiles with windows path separator `\`
causing this conversion to fail, as a path like:
c:\...\zephyr\samples\drivers\led_ws2812\nrf52-bindings.h

resulting in a file list as:
c:;...;zephyr;samples;drivers;led_ws2812;nrf52-bindings.h

which results in a CMake configure dependency to be added to `C:`.
As C: is always newer than the build.ninja file, this resulted in
continues CMake re [... cut]
closed 2021-09-11 by cfriedt