zephyrproject-rtos/zephyr #37639: Module: TF-M: Update to 1.4.0 Release

microbuilder, 2021-08-12:
TF-M 1.4.0 has been released, requiring access to the following commits in related Zephyr modules to update:

- [x] TFM_GIT_SHA  `7ef9178adad866d48e3af42d8a3129dfab792ed8`
- [x] TFM_TESTS_GIT_SHA `e1a8c9fb8394b1f6ea66d2611c070915b0d4b573`
- [x] PSA_ARCH_TESTS_GIT_SHA `51ff2bdfae043f6dd0813b000d928c4bda172660`
- [ ] mbedtls `mbedtls-3.0.0` (`8df2f8e7b9c7bb9390ac74bb7bace27edca81a2b`)
- [ ] mcuboot `TF-Mv1.4-integ` (`4f8091318b4026d14af9e5a7036825bff62fb612`)

The first three are part of the Zephyr's TF-M module, so can be included in a PR, but Zephyr's [mbdtls](https://github.com/zephyrproject-rtos/mbedtls) and [mcuboot](https://github.com/zephyrproject-rtos/mcuboot) modules need to be updated to contain the required commits.

This issue is being filed to track updates to mbedtls and mcuboot.

microbuilder, 2021-08-12:
@d3zd3z Was there any further visibility on the mbed 3.0 update, and perhaps an update to mcuboot to avoid the need to fall back to a TF-M specific fork?

microbuilder, 2021-08-12:
@galak FYI

microbuilder, 2021-08-12:
@galak @ioannisg FYI

ioannisg, 2021-08-16:
Noted this @microbuilder 
Sent a PR for the module update.
We need to fix the mbedtls first, though (possibly the mcuboot as well)

closed 2022-01-18 by microbuilder