zephyrproject-rtos/zephyr #37660: Changing zephyr,console requires a clean build

gmarull, 2021-08-13:
**Describe the bug**

If `zephyr,console` is changed from one device to another without doing a clean build, the console is kept on the first device.

**To Reproduce**

```
west build -b nrf52840dk_nrf52840 samples/hello_world
west flash
```

Observe hello world message on uart0. Change `zephyr,console` to `uart1`, then flash again:

```
west flash
```

Observe hello world message is still displayed on `uart0`.

**Expected behavior**

Changing `zephyr,console` should have an immediate impact, clean build should not be required.

**Impact**

Annoyance, may be a symptom of a bigger problem (not exclusive to `zephyr,console`).

**Logs and console output**

N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: SDK 0.13.0
 - Commit SHA or Version used: a56c42574e3e3bb8b1a35913cc38b21479a858d9

**Additional context**
N/A

tejlmand, 2021-08-13:
some testing reveals that `<build>/zephyr/zephyr.dts` gets updated correctly.

This indicates that the problem is not related to re-generation of devicetree files (headers in particular), but that ninja doesn't know which sources are dependent on the generated devicetree headers, and thus ninja doesn't rebuild those sources.

tejlmand, 2021-09-07:
Been investigating a bit more:
Doing the following works:
```
$ cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -Bbuild samples/hello_world/;ninja -Cbuild flash
# Output on uart0
$ sed 's/zephyr,console\ =\ \&uart0\;/zephyr,console\ =\ \&uart1\;/g' boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts > /tmp/tmp_file
$ mv /tmp/tmp_file boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
$ ninja -Cbuild flash
# CMake re-runs, then ninja invokes the compiler
# No output on uart0, this is now on uart1.
```

But if I instead do:
```
$  cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -Bbuild samples/hello_world/;ninja -Cbuild flash
# Output on uart0
$ echo "/ {
          chosen {
                  zephyr,console = &uart1;
          };
  };" > samples/hello_world/nrf52840dk_nrf52840.overlay
$ ninja -Cbuild flash
ninja: Entering directory `build'
[0/1] Flashing nrf52840dk_nrf52840
-- west flash: rebuilding
ninja: no work to do.
.....
```
because the `samples/hello_world/nrf52840dk_nrf52840.overlay` is a new file unknown to the build system.
This is identical to adding a new source file where user must update a CMakeLists.txt file to include it.

If adding a new file like: `nrf52840dk_nrf52840.overlay` then it must be ensured that CMake is re-invoked.

This can be done manually, running `cmake .` in the build folder or:
- `touch <build>/CMakeCache.txt`
- `touch <sample>/CMakeLists.txt`

and then run `ninja` or `west build` as usual.

We cannot really depend the build system on files not present in the build system.

@gmarull can you confirm that you actually add a new file, and that's why it didn't work ?

gmarull, 2021-09-07:
@tejlmand It looks like I can reproduce the issue on https://github.com/zephyrproject-rtos/zephyr/commit/a56c42574e3e3bb8b1a35913cc38b21479a858d9 but not on `main`, something has happened in between :-)

tejlmand, 2021-09-07:
> @tejlmand It looks like I can reproduce the issue on [a56c425](https://github.com/zephyrproject-rtos/zephyr/commit/a56c42574e3e3bb8b1a35913cc38b21479a858d9) but not on `main`, something has happened in between :-)

This commit fixed the issue:
https://github.com/zephyrproject-rtos/zephyr/commit/e78b0785b3bd20889823bb7edee40d109b28c871

which makes sense because a default value in Kconfig is not updated if the setting is having a prompt.

closed 2021-09-07 by tejlmand