zephyrproject-rtos/zephyr #37672: Board qemu_x86 is no longer working with shell

jakub-uC, 2021-08-13:
Shell does not respond to keyboard input for qemu_x86 board. The same problem does NOT occur for qemu_x86_64.

**To Reproduce**
Steps to reproduce the behavior:
1. Build zephyr/samples/subsys/shell/shell_module
2. west build -b qemu_x86
3. west build -t run
4. See an error

**Expected behavior**
Shell shall respond to keyboard input for qemu_x86 board.

I've made a git bisect and it seems that the following change introduces the problem:

```
eeb15aa3930a2b6c50614f7be13c60b4af52b722 is the first bad commit
commit eeb15aa3930a2b6c50614f7be13c60b4af52b722
Author: Dong Wang <dong.d.wang@intel.com>
Date:   Thu Jul 22 17:56:48 2021 +0800
    timer: hpet: enable 64 bit mode for better usages
    
    Get longer maximum timeout
    Make HPET counter usable as timestamp
    
    Signed-off-by: Dong Wang <dong.d.wang@intel.com>
:040000 040000 9942053d5a4eea7ea549b9f01b3df5c94e8f2693 f1b2d400ee9b873df501018705abedeb818ecdb2 M    drivers
```

dcpleung, 2021-08-16:
@kwd-doodling This is broken after the HPET driver changes.

kwd-doodling, 2021-08-16:
It seems UART interrupt mode goes out of work and not a timer issue. 
by removing "select SERIAL_SUPPORT_INTERRUPT" in drivers/serial/Kconfig.ns16550, switching UART to polling mode, the app recovers  to work. Need more time to debug.

kwd-doodling, 2021-08-18:
could be fix also by:
        -#define HPET_MAX_TICKS ((int32_t)0x7fffffff)
        +#define MAX_TICKS ((int32_t)0x7fffffff)
        +#define HPET_MAX_TICKS \
        +       ((MAX_TICKS - cyc_per_tick) / cyc_per_tick)
cannot understand why a long timeout affects UART function on Qemu.
By the way, it seems Qemu time flies far far quick than real time checking by inputting "kernel uptime" in UART console..

dcpleung, 2021-08-18:
We use QEMU with icount enabled to allow faster testing, where it runs independent of host's clock.

kwd-doodling, 2021-08-23:
with CONFIG_QEMU_ICOUNT=n, the issue is gone, and Qemu time runs about same as real time.
can this be the solution for using shell in Qemu? Who can help on Qemu side?

dcpleung, 2021-08-24:
Using QEMU with icount speeds up CI testing quite a lot, and we really do not want to disable it.

commit cd5357ca (2021-09-15): boards: qemu_x86: disable QEMU icount with shell and HPET timer

For some unknown reasons, using the new modified HPET timer would
render shell unresponsive under QEMU. So for now, disable QEMU
icount when shell is enabled.

Fixes #37672
closed 2021-09-15 by nashif