zephyrproject-rtos/zephyr #37675: tests/kernel/device/kernel.device.pm fails on bt510/bt6x0

galak, 2021-08-13:
`./scripts/twister -s tests/kernel/device/kernel.device.pm -p bt510 -p bt6x0`

gets:

```
[ 77%] Building C object zephyr/drivers/sensor/si7055/CMakeFiles/drivers__sensor__si7055.dir/si7055.c.obj
In file included from /home/galak/git/zephyr/include/kernel_includes.h:21,
                 from /home/galak/git/zephyr/include/kernel.h:17,
                 from /home/galak/git/zephyr/include/init.h:11,
                 from /home/galak/git/zephyr/include/device.h:29,
                 from /home/galak/git/zephyr/include/drivers/sensor.h:23,
                 from /home/galak/git/zephyr/drivers/sensor/si7055/si7055.c:10:
/home/galak/git/zephyr/twister-out/bt510/tests/kernel/device/kernel.device.pm/zephyr/include/generated/devicetree_unfixed.h:7232:35: error: 'DT_N_S_soc_S_i2c_40003000_S_si7055_40_P_wakeup_source' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_i2c_40003000_S_lis2dh12_18_P_wakeup_source'?
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

lairdjm, 2021-08-13:
@galak A bit more information given that the twister tool doesn't run on our development platform (windows) would be nice - what is failing exactly? There is no power management in the Si7055 driver, if the test tool is expecting it then it seems like a bug in the test tool, just like there is no power management in the Si7006 or Si7060 drivers.

galak, 2021-08-13:
We run nightly builds and these boards fail building the test.  If power mgmt isn't supported on the boards than the feature should be disabled or the test excluded for those platforms.

Here are links to the test runs:
https://buildkite.com/zephyr/zephyr-daily/builds/424#92f62569-80b0-4106-bfea-78947595ad64
https://buildkite.com/zephyr/zephyr-daily/builds/424#202597a2-9449-486e-b6b3-1ee67b6bd46d

lairdjm, 2021-08-13:
We didn't change/add any tests or enable/disable anything so I'm not sure how this is our problem? Various Laird Connectivity board files were recently updated to change a Kconfig for the crystal error rate and that's it, which is unrelated to this issue.

galak, 2021-08-13:
This might be a more generic PM issue and not specific to the bt510/bt6x0 boards.

PR #37678 (2021-08-13): dts: bindings: Add missing binding for SI7055 sensor
Add missing dts binding for sensor, this was causing build errors with
CONFIG_PM set on bt510/bt6x0 boards.

Fixes #37675
commit 1deb3cff (2021-08-13): dts: bindings: Add missing binding for SI7055 sensor

Add missing dts binding for sensor, this was causing build errors with
CONFIG_PM set on bt510/bt6x0 boards.

Fixes #37675
closed 2021-08-13 by galak