zephyrproject-rtos/zephyr #37676: tests/kernel/device/kernel.device.pm (and tests/subsys/pm/power_mgmt/subsys.pm.device_pm) fails to build on mec172xevb_assy6906 & mec1501modular_assy6885

galak, 2021-08-13:
`./scripts/twister -s tests/kernel/device/kernel.device.pm -p mec1501modular_assy6885 -p mec172xevb_assy6906`

gets:

```
[ 32%] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/pinmux/pinmux_mchp_xec.c.obj
In file included from /home/galak/git/zephyr/include/kernel_includes.h:21,
                 from /home/galak/git/zephyr/include/kernel.h:17,
                 from /home/galak/git/zephyr/include/init.h:11,
                 from /home/galak/git/zephyr/include/device.h:29,
                 from /home/galak/git/zephyr/drivers/pinmux/pinmux_mchp_xec.c:8:
/home/galak/git/zephyr/twister-out/mec172xevb_assy6906/tests/kernel/device/kernel.device.pm/zephyr/include/generated/devicetree_unfixed.h:1697:56: error: 'DT_N_S_pinmux_S_pinmux_0_P_wakeup_source' undeclared here (not in a function); did you mean 'DT_N_S_cpus_S_cpu_0_P_wakeup_source'?
 1697 | #define DT_N_S_pinmux_FOREACH_CHILD_STATUS_OKAY(fn) fn(DT_N_S_pinmux_S_pinmux_0) fn(DT_N_S_pinmux_S_pinmux_1) fn(DT_N_S_pinmux_S_pinmux_2) fn(DT_N_S_pinmux_S_pinmux_3) fn(DT_N_S_pinmux_S_pinmux_4) fn(DT_N_S_pinmux_S_pinmux_5)
```

scottwcpg, 2021-08-13:
@galak  I read the twitster documentation and tried adding ignore_tags for pm and subsys to the board.  Twister keeps running tests/subsys/pm/device_runtime, tests/subsys/pm/power_mgmt, and tests/kernel.device.pm
Add the following to board yaml:
testing:
  ignore_tags:
   - pm
   - subsys

scottwcpg, 2021-08-13:
@galak Created PR 37681 adding you as a reviewer. I had to add test ignore for device, pm, power, and subsys to prevent all the PM tests from being run.

PR #37679 (2021-08-16): pm: device: Fix if devicetree 'wakeup-source' prop isnt defined
Its possible that a dts binding doesn't inherit from base.yaml and
thus doesn't have `wakeup-source` defined.  To handle these cases
use DT_PROP_OR() which can deal with a property not existing at
all.

Fixes #37676
commit ed76b915 (2021-08-16): pm: device: Fix if devicetree 'wakeup-source' prop isnt defined

Its possible that a dts binding doesn't inherit from base.yaml and
thus doesn't have `wakeup-source` defined.  To handle these cases
use DT_PROP_OR() which can deal with a property not existing at
all.

Fixes #37676
closed 2021-08-16 by cfriedt