zephyrproject-rtos/zephyr #37683: STM32 Eth Tx DMA always uses first descriptor instead of going through circular buffer

fazerlicourice7, 2021-08-13:
**Describe the bug**
The `eth_stm32_hal.c` driver appears to always use the first element of the circular desc / buffer. 

**Expected behavior**
Isn't the point of the circular buffer to enable ~async interaction between cpu and dma controller. I think a pointer should be maintained that gets updated and wraps so we can always point to the next available desc /buffer.  Alternatively, the `ETH_HandleTypeDef` object has enough information to go get the next available desc dynamically.

**Impact**
Just a qualm. Not breaking. I'd call this a performance related issue. 

**Logs and console output**
See [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ethernet/eth_stm32_hal.c#L198):
```
#if defined(CONFIG_SOC_SERIES_STM32H7X)
	const uint32_t cur_tx_desc_idx = 0;  /* heth->TxDescList.CurTxDesc; */
#endif

	dma_tx_desc = GET_FIRST_DMA_TX_DESC(heth);
	while (IS_ETH_DMATXDESC_OWN(dma_tx_desc) != (uint32_t)RESET) {
		k_yield();
	}

#if defined(CONFIG_SOC_SERIES_STM32H7X)
	dma_buffer = dma_tx_buffer[cur_tx_desc_idx];
#else
	dma_buffer = (uint8_t *)(dma_tx_desc->Buffer1Addr);
#endif /* CONFIG_SOC_SERIES_STM32H7X */
```

**Environment:**
 - OS: MacOS
 - Toolchain Zephyr SDK
 - Commit SHA or Version used

**Additional context**
What's most interesting is this line `const uint32_t cur_tx_desc_idx = 0;  /* heth->TxDescList.CurTxDesc; */`  which makes me think that this was attempted properly at some point.

fazerlicourice7, 2021-08-13:
If I'm not missing anything, I can probably take a stab at fixing this.

erwango, 2021-11-08:
>  I can probably take a stab at fixing this.

@fazerlicourice7, Just go ahead if you have some time.

lochej, 2021-11-08:
I remember that I asked myself the same question while working on the ethernet driver. I think there was a few reasons for the conclusion of using just the first DMA buffer to send a packet. I also remember struggling using the circular buffer on F4 and H7 STM32s because handlers and registers are defined differently slightly and that made it tricky. That's true that I tried to implement the circular buffer correctly on top of @Nukersson's H7 implementation but failed.

Maybe I'm wrong but the **send** API function for the ethernet driver is synchronous, meaning that it only returns when 1 buffer with maximum MTU size is sent. Every packet bigger than an Ethernet MTU is disgarded at the **eth_tx** entry.
In fact, if the **send** API should return an error code if something went wrong with the transaction to notify the upper layer. The use of the circular buffer would have been good if multiple ETH frames were to be sent in the same **send** call, so then the DMA setup and send cost would have been factored out for the 2 frames.

My assumption was that since we can only send 1 packet at a time, the implementation was kept rather simplistic.

@fazerlicourice7 But I'm in favor of some refactoring of this part especially if it can improve feature support (like timestamping).👍 But I'm still a bit unsure about performance improvement due to how we the upper layers use the **send** Ethernet API.

Nukersson, 2021-11-08:
There were a new h7 driver merged into main repository, created by baylibre from scratch. It has reduced functionality currently. I think it is much better to follow baylibre's approach.

closed 2022-03-27 by github-actions[bot]