zephyrproject-rtos/zephyr #37684: Add State Machine Framework to Zephyr

sambhurst, 2021-08-13:
## Introduction

Add an application agnostic State Machine Framework to Zephyr that provides an easy way for developers to integrate state machines into their application.

### Problem description

Many developers implement state machines with case statements, if statements, or array function pointers, and these constructs work very well but they don't scale for more complicated state machines. For example, state machines that have tens or more states, states that share behavior, and executing entry and exit actions upon entering and exiting states. Also, it's error pron and time consuming to re-implement state machine constructs for each project. 

### Proposed change

The State Machine Framework lives in the **zephyr/lib/smf/** directory and can be added to any project by including the **CONFIG_SMF=y** into the projects **prj.conf** file. 

## Detailed RFC

#### State Creation
A state is represented by three functions, where one function implements the Entry actions, another function implements the Run actions, and the last function implements the Exit actions. The prototype for these functions is as follows: **void func(void *obj)**, where the **obj** parameter is a user defined structure that has the state machine context, **struct smf_ctx**, as its first member. For example:

```
struct user_object {
    struct smf_ctx ctx;
    /* All User Defined Data Follows */
};
```

The **struct smf_ctx** member must be first because the state machine framework's functions casts the user defined object to the **struct smf_ctx** type with the following macro: **SMF_CTX(o)**, where SMF_CTX is defined as follows:

` #define SMF_CTX(uo) ((struct smf_ctx *)&uo)`

The following macros can be used for easy state creation:

- HSM_CREATE_MSG_STATE
- HSM_CREATE_STATE
- FSM_CREATE_MSG_STATE
- FSM_CREATE_BASIC_STATE

**HSM_CREATE_MSG_STATE(me, mx, e, r, x, p)** is used to create a hierarchical state that can have an optional parent state and prints a message in the optional Entry action and a message in the optional Exit action. The macro has the following parameters:

me - String message that's printed in the Entry action. Can be NULL.
mx - String message that's printed in the Exit action. Can be NULL.
e - Function pointer to state's Entry action. Can be NULL.
r - Function pointer to state's Run action. Can be NULL.
x - Function pointer to state's Exit action. Can be NULL.
p - Function pointer to state's Parent state. Can be NULL.

**HSM_CREATE_STATE(e, r, x, p)** is the same as HSM_CREATE_MSG_STATE except no Entry or Exit messages are printed.

**FSM_CREATE_MSG_STATE(me, mx, e, r, x)** is used to create a flat state and prints a message in the optional Entry action and a message in the optional Exit action. The macro has the following parameters:

me - String message that's printed in the Entry action. Can be NULL.
mx - String message that's printed in the Exit action. Can be NULL.
e - Function pointer to state's Entry action. Can be NULL.
r - Function pointer to state's Run action. Can be NULL.
x - Function pointer to state's Exit action. Can be NULL.

**FSM_CREATE_STATE(e, r, x)** is the same as FSM_CREATE_MSG_STATE except no Entry or Exit messages are printed.

To enable message printing, the **CONFIG_SMF_PRINT_MSG=y** config option must be added to the projects prj.conf file. The use of **HSM_CREATE_MSG_STATE** or **FSM_CREATE_MSG_STATE** when the **CONFIG_SMF_PRINT_MSG** config option isn't defined, results in a build error.

To enable a flat state machine, the **CONFIG_SMF_FLAT=y**  config option must be added to the projects prj.conf file. The use of **FSM_CREATE_MSG_STATE** or **FSM_CREATE_STATE** when the **CONFIG_SMF_FLAT** config option isn't defined, results in a build error.

If the **CONFIG_SMF_FLAT=y** config option is defined and **HSM_CREATE_STATE** or **HSM_CREATE_MSG_STATE** is used, results in a build error.

#### State Machine Creation
A state machine is created by defining a table of states that's indexed by an enum. For example:

```
enum demo_state { S0, S1, S2 };

const struct smf_state demo_states {
	[S0] = FSM_CREATE_STATE(s0_entry, s0_run, s0_exit),
	[S1] = FSM_CREATE_STATE(s1_entry, s1_run, s1_exit),
	[S2] = FSM_CREATE_STATE(s2_entry, s2_run, s2_exit)
};
```

To set the initial state or to transition from one state to another, the **set_state** function is used and it has the following prototype: **void set_state(smf_ctx \*ctx, smf_state \*state)**

NOTE: While the state machine is running, **set_state** should only be called from the Entry and Run functions. Calling **set_state** from the Exit functions doesn't make sense and will generate a warning.

#### State Machine Execution
To run the state machine, the **run_state** function should be called in some application dependent way. For example, a while loop, thread, interrupt, etc.. The function has the following prototype: **void run_state(smf_ctx \*ctx)**

### Proposed change (Detailed)

No more detail here.

### Dependencies

No dependencies

### Concerns and Unresolved Questions

No concerns

## Alternatives

Initially, each state consisted of one function instead of three and used a local case statement to implement entry, run, and exit actions but this implementation resulted in a cluttered state.

hongshui3000, 2021-08-14:
This has always been the feature I want, but there seems to be not many people interested in the community

carlescufi, 2021-08-16:
@sambhurst this is a reasonable idea. I think one of the hardest parts is going to be agreeing to a common API for the subsystem that covers the needs of all state machines in the tree so that they too can transition to the new framework. In particular I think this should be able to run on both threads and interrupt context.

sambhurst, 2021-08-24:
The API mainly consists of set_state and run_state. Also, these function are interrupt and thread safe. Of course someone could pass the same state machine context to separate threads and I don't know why someone would but if they did, proper synchronization would have to be used.

sambhurst, 2021-08-29:
I have taken the liberty of generating a pull request for this issue. To add more context, this framework is part of a larger effort of porting the Google ChromeOS USB-C framework to Zephyr and an issue for that port will be come soon.
https://github.com/sambhurst/zephyr/pull/new/add_smf_issue_37684

hongshui3000, 2021-08-30:
@sambhurst  I take the liberty to ask, does the framework have to use multi-threading? Can I use it in a single-threaded environment?

sambhurst, 2021-08-30:
I apologize, I thought I answered these questions in a previous reply. The framework is thread safe; no shared resources are accessed by the API. The only data structure used by the framework is the state machine context and each thread should maintain it's own state machine context.