zephyrproject-rtos/zephyr #37696: Modbus TCP: wrong transaction id in response

ArseniSoitu, 2021-08-14:
Hello.
Tried to use Modbus in TCP mode and found, that reply from server has wrong transaction id in response packet.

Found code where response header prepares and it seems that transaction id needs to be assigned after function code:
`ctx->tx_adu.trans_id = ctx->rx_adu.trans_id;`

https://github.com/zephyrproject-rtos/zephyr/blob/f3435b75efbfd7614ff873f8bfa90a031d6d27e8/subsys/modbus/modbus_server.c#L954

stephanosio, 2021-08-15:
cc @jfischer-no

carlescufi, 2021-08-16:
@ArseniSoitu would you mind sending a PR to fix this?

jfischer-no, 2021-08-19:
I do not see the bug, also modbus_server.c should not be changed for that, your app should implement raw_tx_ct and copy the trans_id there, see https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/modbus/tcp_server/src/main.c#L126.

jfischer-no, 2021-09-15:
resolved by 19c519eb05d6d24a23fd084a3ab6c1a78edf5536

closed 2021-09-15 by jfischer-no