zephyrproject-rtos/zephyr #37701: stm32: CONFIG_PRINTK=y conflicts with uart serial DMA

manoj153, 2021-08-15:
**Describe the bug**
When `CONFIG_PRINTK=y` throws erros

**To Reproduce**
Steps to reproduce the behavior:
1. Build the code

**Expected behavior**
Build successfully 

**Impact**
Could not build the code. 

**Logs and console output**
```
t_stm32.c.obj -c /home/manoj/zephyrproject/zephyr/drivers/serial/uart_stm32.c
In file included from /home/manoj/zephyrproject/zephyr/include/sys/util_macro.h:34,
                 from /home/manoj/zephyrproject/zephyr/include/sys/util.h:17,
                 from /home/manoj/zephyrproject/zephyr/include/kernel/sched_priq.h:9,
                 from /home/manoj/zephyrproject/zephyr/include/kernel_includes.h:23,
                 from /home/manoj/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/manoj/zephyrproject/zephyr/drivers/serial/uart_stm32.c:17:
zephyr/include/generated/devicetree_unfixed.h:9613:36: error: 'DT_N_S_soc_S_serial_40004800_P_dmas_NAME_rx_VAL_features' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_serial_40004800_P_dmas_NAME_rx_VAL_channel'?
 9613 | #define DT_N_INST_1_st_stm32_uart  DT_N_S_soc_S_serial_40004800
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/manoj/zephyrproject/zephyr/include/sys/util_internal.h:67:26: note: in definition of macro '__DEBRACKET'
   67 | #define __DEBRACKET(...) __VA_ARGS__
```

FRASTM, 2021-08-16:
@manoj153 thanks for reporting this issue. Is a particular target causing the error ?

manoj153, 2021-08-16:
> @manoj153 thanks for reporting this issue. Is a particular target causing the error ?

tried with `nucleo_h743zi` and `nucleo_h723zg` shows similar error

FRASTM, 2021-08-16:
I am trying with another instance of uart (UART2), as the uart3 is for the console

manoj153, 2021-08-16:
> I am trying with another instance of uart (UART2), as the uart3 is for the console

Let me know if it's at least reproducible at your end..

FRASTM, 2021-08-16:
Yes, there are several pb with this platform using the DMA over a peripheral, 
because the type of DMA is V1 with a DMAMUX.
 Even if the feature in the dma-cell exist, because of the DMAMUX it should not appear.
The stream offset is also initialized like it is with the DMAMUX.
The USART2 is choosen for the test, pins are availbale on the CN9 connector PD5, PD6.
This platform which has Data cache is suitable when DMA is on, thats the reason for a CONFIG_NOCACHE_MEMORY flag to be set (like stm32f7 family).

roland-xz, 2021-08-17:
Having the same/similar problem, on STM32F105.
**DT config:**
```
&usart1 {
	pinctrl-0 = <&usart1_tx_pa9 &usart1_rx_pa10>;
	current-speed = <230400>;
	status = "okay";
	dmas = <&dma1 4 0x440>, <&dma1 5 0x480>;
	dma-names = "tx", "rx";	
};
```

**Error log:**
```
In file included from ../zephyr/include/sys/util_macro.h:34,
                 from ../zephyr/include/sys/util.h:17,
                 from ../zephyr/include/kernel/sched_priq.h:9,
                 from ../zephyr/include/kernel_includes.h:23,
                 from ../zephyr/include/kernel.h:17,
                 from ../zephyr/drivers/serial/uart_stm32.c:17:
zephyr/include/generated/devicetree_unfixed.h:4910:36: error: 'DT_N_S_soc_S_serial_40013800_P_dmas_NAME_rx_VAL_slot' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_serial_40013800_P_dmas_NAME_rx_VAL_channel'?
 4910 | #define DT_N_INST_0_st_stm32_uart  DT_N_S_soc_S_serial_40013800
```

closed 2021-08-20 by cfriedt