zephyrproject-rtos/zephyr #37704: hello world doesn't work on qemu_arc_em when CONFIG_ISR_STACK_SIZE=1048510

wentongwu, 2021-08-16:
**Describe the bug**
hello world sample doesn't work on qemu_arc_em when CONFIG_ISR_STACK_SIZE=1048510

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=qemu_arc_em ..
3. make run
4. See error

wentongwu, 2021-08-16:
The SDK is 0.13.0

IRISZZW, 2021-08-16:
@wentongwu I can reproduce it with qemu_arc_em and nsim_em boards. 
Then I try to CONFIG_ISR_STACK_SIZE=1048576 (1M), it works fine for both boards. I will continue debug it.
I'm so curios how did you get the ``magic number`` 1048510, Could you provide more details?

IRISZZW, 2021-08-31:
@wentongwu  Could you please provide more details?

wentongwu, 2021-08-31:
> @wentongwu Could you please provide more details?

someone told me it's just by accidental.

BTW I think this is a high priority bug because it impacts the basic function of Zephyr OS.

IRISZZW, 2021-09-01:
@wentongwu usually stack address and stack size need stack alignment, 1048510 is not aligned to 4. so it may cause `Misaligned data access` exception.

cfriedt, 2021-09-23:
Is a 1M stack common for ARC? Typically, we see stack sizes of 1kB, 2kB... even Linux has just started using a stack size of 8kB now for x86_64.

@wentongwu - have you tried a more conventional stack size such as `CONFIG_ISR_STACK_SIZE=4096`?

wentongwu, 2021-09-24:
@cfriedt AFAIK nobody uses so large stack size in real product. But I don't want to talk about why we use so large stack in QEMU. And also Linux...

But the problem here is that the none-4byte-align stack size will cause qemu arc run incorrectly. Not sure if you have tried other size like `cmake -DBOARD=qemu_arc_em -DCONFIG_ISR_STACK_SIZE=4094 ..` or other platform like `mps2_an385`, `qemu_x86`, or zephyr old version(like 2.5).

evgeniy-paltsev, 2021-11-24:
It's enough to use any unaligned ISR_STACK_SIZE, I.e ISR_STACK_SIZE=2050 instead of 2048.

The root cause seems to be in how we setup stack pointer in reset.S - we are setting it to unaligned value.

```
#define INIT_STACK z_interrupt_stacks
#define INIT_STACK_SIZE CONFIG_ISR_STACK_SIZE

	mov_s sp, INIT_STACK
	add sp, sp, INIT_STACK_SIZE

``` 

I guess it would be enough to check in compile time that we have proper CONFIG_ISR_STACK_SIZE alignment. 
@ruuddw @abrodkin any objections on this?

abrodkin, 2021-11-24:
Indeed, we shouldn't set bogus values in `SP`. Another question is how to handle those cases. I'd say compile-time warning is OK.
We may otherwise try to compensate for funny sizes in-place, but that may unexpectedly move stack boundaries if we just silently align provided random value. Ah, and BTW what about `z_interrupt_stacks`, are we sure it's properly aligned as well?

ruuddw, 2021-11-24:
Compile-time error seems fine in this case. For user stacks the alignment I believe is handled by macro's rounding up, but the alignment can be different depending on configuration (e.g. with or without MPU). In this case the alignment requirement is always the same, and word aligned stack and stack size are reasonable to enforce.

PR #49063 (2022-08-16): ARC: Check that ISR and exception stacks are proper aligned
`CONFIG_ISR_STACK_SIZE` and `CONFIG_ARC_EXCEPTION_STACK_SIZE` are used in arc asembler code without C wrapper on them with following manner:

```
#define INIT_STACK z_interrupt_stacks
#define INIT_STACK_SIZE CONFIG_ISR_STACK_SIZE

	mov_s sp, INIT_STACK
	add sp, sp, INIT_STACK_SIZE
```

In result we can get stack unaligned if we have `CONFIG_ISR_STACK_SIZE` or `CONFIG_ARC_EXCEPTION_STACK_SIZE`
values unaligned. 

There is no point to have unaligned stack size - so let's check against such configurations in compile time.

Fixes: #37704
closed 2022-08-16 by fabiobaltieri