zephyrproject-rtos/zephyr #37709: x86 PCIe ECAM does not work as expected

tbursztyka, 2021-08-16:
**Describe the bug**
PCIe ECAM "Extended Configuration Access Mechanism" based on MMIO is supported on x86. 
However, until now, accessing the extended configuration space has never been tried nor was useful.
But since pr #37529 it seems that such access is not functioning as expected. It does work fine on the legacy configuration space, but not the extended one.

**To Reproduce**

Steps to reproduce the behavior:
1. Get the mentioned PR above, and get to the shell_module sample
2. mkdir build; cd build
3. cmake -DBOARD=ehl_crb
4. Apply this quick patch on  arch/x86/core/acpi.c  (ACPI is not working on EHL, the only way to get it to work is to improve EFI support)
```
-       search_phys = 0xe0000;
+       search_phys = 0x7ac4e014;
```
5. make
6. Run the zephyr.efi on EHL target
7. Run command "pcie"

**Expected behavior**
PCIe device 0:1c:0 must show extended capabilities available.
It supports ACS and PTM (that last one needs to be enabled in the BIOS however).

On the contrary, PCIe device 0:0:0 must not show such availabality

**Impact**
Extended Configuration Area is not accessible, or at least not reliably.
This prevent future features to be supported atm.

**Logs and console output**
(Some extra output where added to the pcie shell module)
```
0:0.0 ID 8086:4518 class 6 subclass 0 prog i/f 0 rev 1
    wired interrupt on IRQ 0
    Extended Capabilities available (0xffffffff)
PTM? : 0xffffffff

(...)

0:1c.0 ID 8086:4b38 class 6 subclass 4 prog i/f 0 rev 11 [bridge]
    No Extended Capabilities available (0x0)
PTM? : 0x0
```

See the 0xffffffff ? this should NOT happen. It should be valid data or 0x0.
On the contrary, on 0:1c:0 it shows 0x0 but this is bogus. It must show some valid content.

To verify that, reboot the EHL target, go to the EFI shell and run: 
```
shell> pci 0 1c 0 -i
(...)
Start dumping PCIex extended configuration space (0x100 - 0xFFF).

  00000100: 01 00 01 22 00 00 00 00-00 00 00 00 11 00 06 00  *..."............*
  00000110: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000120: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
(...)
```

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK
 - Latest mastel

dcpleung, 2021-08-17:
If you change the flags in `z_phys_map()` from `0` to `K_MEM_CACHE_NONE`, would it work? The default is write-back, which might be the issue.

tbursztyka, 2021-08-18:
@dcpleung https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/x86/core/pcie.c#L54 it's already the case. Or do you mean elsewhere?

dcpleung, 2021-08-18:
The one that maps `0xe0000` (in your case, `0x7ac4e014`) does not have the `K_MEM_CACHE_NONE` flag. Wonder if that helps?

tbursztyka, 2021-08-20:
Don't bother looking at this issue. It's not an issue. Or to be honest: the thing that wrote pr #37529 is the issue. 

(Completely forgot that pcie_conf_reg actually works by 32bits, so to reach offset 0x100 reg should be 0x100/4)

closed 2021-08-20 by tbursztyka