zephyrproject-rtos/zephyr #37710: Bluetooth advert packet size is size of maximum packet not size of actual data

thedjnK, 2021-08-16:
**Describe the bug**
Data is being added to an advert and scan response when Bluetooth advertising is starting on an nRF5340 with a single 128-bit UUID and short 25 character device name, this results in a packet of 259 bytes being sent to the network core even though the data is no-where near this long. The problem in the code, subsys\bluetooth\host\adv.c seems to be this:

```
/* Set legacy data using Extended Advertising HCI commands */
static int hci_set_ad_ext(struct bt_le_ext_adv *adv, uint16_t hci_op,
			  const struct bt_ad *ad, size_t ad_len)
{
	struct bt_hci_cp_le_set_ext_adv_data *set_data;
	struct net_buf *buf;
	int err;

	buf = bt_hci_cmd_create(hci_op, sizeof(*set_data));
	if (!buf) {
		return -ENOBUFS;
	}

	set_data = net_buf_add(buf, sizeof(*set_data));
	(void)memset(set_data, 0, sizeof(*set_data));

	err = set_data_add(set_data->data, BT_HCI_LE_EXT_ADV_FRAG_MAX_LEN,
			   ad, ad_len, &set_data->len);
	if (err) {
		net_buf_unref(buf);
		return err;
	}

	set_data->handle = adv->handle;
	set_data->op = BT_HCI_LE_EXT_ADV_OP_COMPLETE_DATA;
	set_data->frag_pref = BT_HCI_LE_EXT_ADV_FRAG_DISABLED;

	return bt_hci_cmd_send_sync(hci_op, buf, NULL);
}
```
Here, sizeof(*set_data) is 255 bytes and this full buffer is passed via HCI to the other side rather than the actual size of the data in the buffer

**Impact**
Annoyance, it can be worked around by allowing a larger buffer size on the network core application but is inefficient.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: GNU Arm Embedded Toolchain 10.3-2021.07
 - Commit SHA or Version used: nRF Connect SDK 1.6.1 which uses Zephyr 2.6

joerchan, 2021-08-16:
Note: This would not be only advertising commands, but also other commands that are of variable length.

Thalley, 2021-08-17:
I wonder if "bug" is a proper label for this, as it more of an optimization request, is it not?

joerchan, 2021-08-17:
I think we can call it that. Actually I'm not sure what the impact on performance is here. If RPMsg is using shared memory then nothing should be "sent" in this case.
But it would be interested to know the impact of HCI over UART as well.

thedjnK, 2021-08-17:
The reason I labelled this a bug was because we were getting advertising failed errors in our application when built for the nRF5340 but they worked fine on the nRF52840, our adverts aren't large in size and so we have to specifically add a workaround for the nRF5340 for this issue to support the large message

joerchan, 2021-08-18:
@thedjnK Enabling `CONFIG_BT_EXT_ADV=y` on hci_rpmsg would make the buffers equal size on both sides.

PR #43255 (2022-03-09): Bluetooth: Host: Optimize hci buf alloc by calc actually require size
By calculating the actual space required instead of using the maximum space, 
for `ad_stream`, add two related APIs for save and restore status to calculate the total space.

Fixes #37710
closed 2022-03-09 by nashif