zephyrproject-rtos/zephyr #37730: http_client_req: Timeout likely not working as expected

ycsin, 2021-08-17:
**Describe the bug**
The `hawkbit` subsystem is using the http client library to download firmware. However I've never seen that a timeout occurred when the download stops. I'm using a Quectel EC21 modem with the `gsm_ppp` driver.

**To Reproduce**
I'm not sure how to manually stop the download in hawkbit, maybe turn off the modem manually during the download?

**Expected behavior**
I expect the request to timeout after the defined duration.

**Impact**
Application that uses the function call might not return from it.

**Logs and console output**
[http_client timeout.txt](https://github.com/zephyrproject-rtos/zephyr/files/6997833/http_client.timeout.txt)

**Environment (please complete the following information):**
 - OS: Win 10
 - Toolchain gnuarmemb
 - v2.6.0-1638-g0296d12675fd

ycsin, 2021-08-19:
@cfriedt I feel like this is more of an issue of the `http_client` rather than a `hawkbit` ones

mniestroj, 2021-08-20:
@ycsin How did you configure this timeout? When should it expire?

mniestroj, 2021-08-23:
I confirm that there is problem with timeout handling inside http client. Currently `close()` is called on the underlying file descriptor when timeout happens, which does not wakeup `recv()` thread. I guess that waking up did work some time ago, before `k_condvar` was used for waiting on incoming data. Calling `(void)k_condvar_signal(&ctx->cond.recv);` from `close()` implementation seems to help (to wakeup `recv()` thread), but I am not sure we should actually do that. Looking at `close()` manual page, there is a note:
```
   Multithreaded processes and close()
       It is probably unwise to close file descriptors while they  may  be  in
       use by system calls in other threads in the same process.  Since a file
       descriptor may be reused, there are some obscure race  conditions  that
       may cause unintended side effects.

       Furthermore, consider the following scenario where two threads are per‐
       forming operations on the same file descriptor:

       1. One thread is blocked in an I/O system call on the file  descriptor.
          For  example,  it  is  trying  to write(2) to a pipe that is already
          full, or trying to read(2) from a stream socket which currently  has
          no available data.

       2. Another thread closes the file descriptor.

       The behavior in this situation varies across systems.  On some systems,
       when the file descriptor is closed, the blocking  system  call  returns
       immediately with an error.

       On  Linux (and possibly some other systems), the behavior is different:
       the blocking I/O system call holds a reference to the  underlying  open
       file  description,  and this reference keeps the description open until
       the I/O system call completes.  (See open(2) for a discussion  of  open
       file descriptions.)  Thus, the blocking system call in the first thread
       may successfully complete after the close() in the second thread.
```

So I think that one way to go is to implement `shutdown()` syscall, which is for now a noop in Zephyr. Then we can call `shutdown(fd, SHUT_RD);` from within http client timeout handler. `shutdown()` implementation should wakeup `recv()` caller, by calling `(void)k_condvar_signal(&ctx->cond.recv);` (and other stuff).

@rlubos we have actually similar problem with calling `close()` by internal library implementation in MQTT client, when one of the threads uses `poll()` on underlying socket and other thread is calling `mqtt_publish()` (which fails and closes file descriptor). IMO we should leave socket close operation to library user (which can be wrapped in `mqtt_disconnect()` and `mqtt_abort()`) and use `shutdown()` (once implemented) in MQTT internal implementation.

rlubos, 2021-09-08:
> I guess that waking up did work some time ago, before k_condvar was used for waiting on incoming data.

Yes, closing a socket caused the blocked threads to unblock immediately, it was specifically implemented that way: https://github.com/zephyrproject-rtos/zephyr/commit/9975f7b85451d62b06a6636eba8830af607fc853#
However, you are right this could cause race conditions if the underlying `net_contex` or sockfd was reused before the blocked thread could run. 

> we have actually similar problem with calling close() by internal library implementation in MQTT client, when one of the threads uses poll() on underlying socket and other thread is calling [...]

This sounds like a reasonable way to handle this. I'm just slightly concerned about the offloaded impelmentation that do not support the feature (at least nRF91 does not support shutdown), but I guess it  shouldn't be a big deal to simulate the behaviour at the glue layer.

Btw, did you find any resources on how should `poll()`/`recv()` behave when `shutdown()` on RD/WR is issued? I didn't find anything useful in the man pages. I'm specifically wondering whether `poll()` shall return any particular flag.

mniestroj, 2021-09-09:
>I'm just slightly concerned about the offloaded impelmentation that do not support the feature (at least nRF91 does not support shutdown), but I guess it shouldn't be a big deal to simulate the behaviour at the glue layer.

There are plenty of offloaded drivers in the Zephyr tree already and none of them support shutdown() currently. For now, for HTTP we can print an error log in timeout handler (https://github.com/zephyrproject-rtos/zephyr/pull/38034/commits/5c4b6093b969afc3ae6d6e61ec6e9ccb1a60144d) when shutdown() fails because it is not supported. This will at least notify HTTP library users why haven't the timeout expired correctly. Maybe something similar could be done for MQTT.

As you say, it might be feasible to handle shutdown() for offloaded drivers in Zephyr sockets implementation. The thing with nrf91 which I don't know how to handle is how to inform nrf91 offloaded poll() that it should return. AFAIK it only returns on incoming data and/or timeout, but not with any other event (e.g. eventfd notification). This is related to https://devzone.nordicsemi.com/f/nordic-q-a/76379/nrf9160-offloaded-poll-is-incompatible-with-other-zephyr-file-descriptors-e-g-eventfd.

>Btw, did you find any resources on how should poll()/recv() behave when shutdown() on RD/WR is issued? I didn't find anything useful in the man pages. I'm specifically wondering whether poll() shall return any particular flag.

No, I haven't found any specific documentation on how `recv()` and `poll()` should behave. I've tested however how does `recv()` behave on Linux and implemented the same behavior. I guess I should do the same for `poll()` (test it on Linux) and then implement unit tests of "shutdown() while poll()" in unit tests.

ycsin, 2022-01-11:
@rlubos @mniestroj 
I wonder if this issue originates from socket or http client? If it originates from the socket API, wouldn't it affect other things, like the MQTT client?

closed 2022-03-02 by mbolivar-nordic