zephyrproject-rtos/zephyr #37734: xtensa xcc build spi_nor.c fail

hongshui3000, 2021-08-17:
**Describe the bug**
'''
FAILED: zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/spi_nor.c.obj 
C:\usr\xtensa\XtDevTools\install\tools\RI-2021.6-win32\XtensaTools\bin\xt-xcc.exe -DBUILD_VERSION=zephyr-v2.6.0-641-gcc0abee3db73 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -IE:/work/ZC3827/ZCAN-IOT/zephyr/include -Izephyr/include/generated 
-IE:/work/soc/xtensa/ -IE:/work/k/soc/xtens/include -IE:/work/include -IE:/work/modules/hal/xtensa/include -IE:/work/ZC3827/ZCAN-IOT/modules/hal/xtensa/zephyr/soc/ -isystem E:/work/zephyr/lib/libc/minimal/include -Os -imacros E:/work/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacrosE:/work/zephyr/include/toolchain/xcc_missing_defs.h -fms-extensions -imacros E:/work/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Werror=implicit-int -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/spi_nor.c.obj -MF zephyr\drivers\flash\CMakeFiles\drivers__flash.dir\spi_nor.c.obj.d -o zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/spi_nor.c.obj -c E:/work/zephyr/drivers/flash/spi_nor.c
Signal: caughtError: Signal caught in phase CGPREP -- processing aborted
 in CGPREP phase.
xt-xcc.exe ERROR:  C:/usr/xtensa/XtDevTools/install/tools/RI-2021.6-win32/XtensaTools/libexec/xcc/be.exe returned non-zero status 3  
[53/87] Building C object zephyr/kernel/CMakeFiles/kernel.dir/errno.c.obj
ninja: build stopped: subcommand failed.
'''
function:
```
static int spi_nor_erase(const struct device *dev, off_t addr, size_t size)
{
	const size_t flash_size = dev_flash_size(dev);
	int ret = 0;

	/* erase area must be subregion of device */
	if ((addr < 0) || ((size + addr) > flash_size)) {
		return -ENODEV;
	}

	/* address must be sector-aligned */
	if (!SPI_NOR_IS_SECTOR_ALIGNED(addr)) {
		return -EINVAL;
	}

	/* size must be a multiple of sectors */
	if ((size % SPI_NOR_SECTOR_SIZE) != 0) {
		return -EINVAL;
	}

	acquire_device(dev);
	ret = spi_nor_write_protection_set(dev, false);

	while ((size > 0) && (ret == 0)) {
		spi_nor_cmd_write(dev, SPI_NOR_CMD_WREN);

		if (size == flash_size) {
			/* chip erase */
			spi_nor_cmd_write(dev, SPI_NOR_CMD_CE);
			size -= flash_size;
		} else {
			const struct jesd216_erase_type *erase_types =
				dev_erase_types(dev);
			const struct jesd216_erase_type *bet = NULL;

			for (uint8_t ei = 0; ei < JESD216_NUM_ERASE_TYPES; ++ei) {
				const struct jesd216_erase_type *etp =
					&erase_types[ei];

				if ((etp->exp != 0)
				    && SPI_NOR_IS_ALIGNED(addr, etp->exp)
				    && SPI_NOR_IS_ALIGNED(size, etp->exp)
				    && ((bet == NULL)
					|| (etp->exp > bet->exp))) {
					bet = etp;
				}
			}
			if (bet != NULL) {
				spi_nor_cmd_addr_write(dev, bet->cmd, addr, NULL, 0);
				addr += BIT(bet->exp);
				size -= BIT(bet->exp);
			} else {
				LOG_DBG("Can't erase %zu at 0x%lx",
					size, (long)addr);
				ret = -EINVAL;
			}
		}
		spi_nor_wait_until_ready(dev); ------------------------------------------------------->here
	}

	int ret2 = spi_nor_write_protection_set(dev, true);

	if (!ret) {
		ret = ret2;
	}

	release_device(dev);

	return ret;
}
```
my fix:
```
static int spi_nor_erase(const struct device *dev, off_t addr, size_t size)
{
	const size_t flash_size = dev_flash_size(dev);
	int ret = 0;

	/* erase area must be subregion of device */
	if ((addr < 0) || ((size + addr) > flash_size)) {
		return -ENODEV;
	}

	/* address must be sector-aligned */
	if (!SPI_NOR_IS_SECTOR_ALIGNED(addr)) {
		return -EINVAL;
	}

	/* size must be a multiple of sectors */
	if ((size % SPI_NOR_SECTOR_SIZE) != 0) {
		return -EINVAL;
	}

	acquire_device(dev);
	ret = spi_nor_write_protection_set(dev, false);

	while ((size > 0) && (ret == 0)) {
		spi_nor_cmd_write(dev, SPI_NOR_CMD_WREN);

		if (size == flash_size) {
			/* chip erase */
			spi_nor_cmd_write(dev, SPI_NOR_CMD_CE);
			size -= flash_size;
		} else {
			const struct jesd216_erase_type *erase_types =
				dev_erase_types(dev);
			const struct jesd216_erase_type *bet = NULL;

			for (uint8_t ei = 0; ei < JESD216_NUM_ERASE_TYPES; ++ei) {
				const struct jesd216_erase_type *etp =
					&erase_types[ei];

				if ((etp->exp != 0)
				    && SPI_NOR_IS_ALIGNED(addr, etp->exp)
				    && SPI_NOR_IS_ALIGNED(size, etp->exp)
				    && ((bet == NULL)
					|| (etp->exp > bet->exp))) {
					bet = etp;
				}
			}
			if (bet != NULL) {
				spi_nor_cmd_addr_write(dev, bet->cmd, addr, NULL, 0);
				addr += BIT(bet->exp);
				size -= BIT(bet->exp);
			} else {
				LOG_DBG("Can't erase %zu at 0x%lx",
					size, (long)addr);
				ret = -EINVAL;
			}
		}
		ret  = spi_nor_wait_until_ready(dev); ----------------------------------------------->
	}

	int ret2 = spi_nor_write_protection_set(dev, true);

	if (!ret) {
		ret = ret2;
	}

	release_device(dev);

	return ret;
}
```

Actually I don’t understand why there is such a issue, but my compiler does have such a issue.
I feel it is a side effect of code optimization.

sylvioalves, 2021-08-17:
@cfriedt could you assign to proper spi_nor owner or maintainer if any?

cfriedt, 2021-08-17:
@hongshui3000 - since you have a fix for this, can you please submit a PR? Would likely make it easier to move along in that case.

Please ensure that your commit message includes `Fixes #37737`.

cfriedt, 2021-08-17:
I think if the issue is toolchain-related, the PR might illustrate that the fix is straightforward, which will encourage more reviewers to accept it.

PR #38045 (2021-08-31): flash: spi_nor: fix building on XCC
For some reason, XCC fails to build complaining segfault
during CGPREP phase. Adding an assignment to a volatile
return value seems to fix this. This provides an easily
revertable commit to workaround the issue.

Fixes #37734
commit ef795f6f (2021-08-31): flash: spi_nor: fix building on XCC

For some reason, XCC fails to build complaining segfault
during CGPREP phase. Adding an assignment to a volatile
return value seems to fix this. This provides an easily
revertable commit to workaround the issue.

Fixes #37734
closed 2021-08-31 by nashif