zephyrproject-rtos/zephyr #37735: Unsigned types are incorrectly serialized when TLV format is used in LWM2M response

icblnk, 2021-08-17:
**Describe the bug**
As LWM2M v1.0.2 supports only signed integers, when Zephyr's LWM2M stack is using `lwm2m+tlv` COAP payload, unsigned integers are incorrectly serialized if, for example, uint8_t value > INT8_MAX.

**To Reproduce**
Steps to reproduce the behavior:
1. Define object field as:
```c
struct lwm2m_engine_obj_field fields[] = {
	OBJ_FIELD_DATA(7, RW, U8)
};
```
2. Set value to `129`
3. Read value back. Currently the value that's sent back is encoded as `c1 07 81`, which is TLV integer representation of `-127`.

**Expected behavior**
Value sent back should be `c2 07 00 81` which is TLV integer representation of `129`.

**Impact**
Zephyr LWM2M stack cannot properly interact with `EMQ X Broker` (see https://github.com/emqx/emqx/issues/4989 for details).

A traffic dump showing the issue is in the attachment.
[signed-integer-tlv.pcap.zip](https://github.com/zephyrproject-rtos/zephyr/files/6998771/signed-integer-tlv.pcap.zip)

rlubos, 2021-08-20:
@icblnk Thanks for reporting, indeed the unsigned integers were casted incorrectly, effectively narrowing their scope. Please have a look at https://github.com/zephyrproject-rtos/zephyr/pull/37845 where I've proposed a fix for the issue.

icblnk, 2021-08-23:
@rlubos thanks for a quick response! The change looks good to me, but I'm wondering what's with `LWM2M_RES_TYPE_U64` case? Should en error be returned in case a value > INT64_MAX?

rlubos, 2021-09-07:
@icblnk Sorry for the late reply, I was on vacation. Yes, I think it makes sense to return an error in this case, I'll update the PR.

rlubos, 2021-09-07:
Actually, after closer look into the spec and code, I'm leaning towards removing `LWM2M_RES_TYPE_U64` entirely. I don't see an error code in the spec that could be returned to the server in case the value could not be encoded properly which could lead to ambiguity. 
Since the LwM2M spec limits the TLV to 8 bytes, making it impossible to encode the full range of `uint64_t` I'm assuming it's not expecting to use such type at all. I'll add a commit that removes the type and replaces existing uses with S64, the're not many. We can discuss further in the PR if this could be approach any other way.

PR #37845 (2021-09-10): net: lwm2m: Fix unsigned integers ecoding in TLV
As the resource values represented by the unsigned integers were casted
to integers of a corresponding size in the read handler, they were not
ecoded properly if the unsigned value was larger than the maximum
integer value of the corresponding size (i.e. they were encoded as
negative values).

Fix this by casting the unsinged value to a wider integer type, to
prevent incorrect interpratetion of the data provided. The TLV encoding
functions take care of the optimization (i. e. encoding integers on the
minimum number of bytes needed), so it should [... cut]
closed 2021-09-10 by carlescufi