zephyrproject-rtos/zephyr #37746: qemu_x86_64 fails samples/hello_world/sample.basic.helloworld.uefi in CI

galak, 2021-08-17:
When trying to run in CI for nightly we get:

```

CMake Error at ../../cmake/emu/qemu.cmake:29 (find_program):
  Could not find UEFI using the following names: uefi-run
Call Stack (most recent call first):
  ../../CMakeLists.txt:1591 (include)

```

Full log here:

https://buildkite.com/zephyr/zephyr-daily/builds/428#577bd93e-e116-4f7c-b25b-c0bd4db17200

galak, 2021-08-17:
Closing issue for now, I think we just need to update CI for daily to docker image 0.18.2:

https://github.com/zephyrproject-rtos/zephyr/pull/37747

galak, 2021-08-18:
Was hoping v0.18.2 docker image would fix the issue, but still seeing it in the nightly build:

https://buildkite.com/zephyr/zephyr-daily/builds/429#66c32748-1b3a-43b8-8a28-40ca9672f3b5

chen-png, 2021-08-19:
@galak  I tested with v0.18.2 docker image on my NUC, it work well. But I found that uefi-run is installed into /root/.cargo/bin directory, so I think this maybe related to the user account used to run docker, maybe I can change a installation directory for uefi-run

chen-png, 2021-09-02:
hi @galak  where could I check the CI results? I am curious why this issue is still existed, but it worked well on my local.

closed 2021-09-02 by galak