zephyrproject-rtos/zephyr #37768: tests/lib/ringbuffer/libraries.data_structures fails to build on mec172xevb_assy6906

galak, 2021-08-18:
`/scripts/twister -s tests/lib/ringbuffer/libraries.data_structures -p mec172xevb_assy6906`

gets:

```
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c: In function 'sys_clock_set_timeout':
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:85:30: error: division by zero [-Werror=div-by-zero]
   85 | #define MAX_TICKS (TIMER_MAX / CYCLES_PER_TICK)
      |                              ^
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:205:44: note: in expansion of macro 'MAX_TICKS'
  205 |  } else if ((n == K_TICKS_FOREVER) || (n > MAX_TICKS)) {
      |                                            ^~~~~~~~~
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:85:30: error: division by zero [-Werror=div-by-zero]
   85 | #define MAX_TICKS (TIMER_MAX / CYCLES_PER_TICK)
      |                              ^
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:206:16: note: in expansion of macro 'MAX_TICKS'
  206 |   full_ticks = MAX_TICKS - 1;
      |                ^~~~~~~~~
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:227:51: error: division by zero [-Werror=div-by-zero]
  227 |  partial_cycles = CYCLES_PER_TICK - (total_cycles % CYCLES_PER_TICK);
      |                                                   ^
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c: In function 'sys_clock_elapsed':
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:262:8: error: division by zero [-Werror=div-by-zero]
  262 |  ticks /= CYCLES_PER_TICK;
      |        ^~
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c: In function 'xec_rtos_timer_isr':
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:85:30: error: division by zero [-Werror=div-by-zero]
   85 | #define MAX_TICKS (TIMER_MAX / CYCLES_PER_TICK)
      |                              ^
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:282:16: note: in expansion of macro 'MAX_TICKS'
  282 |  timer_restart(MAX_TICKS * CYCLES_PER_TICK);
      |                ^~~~~~~~~
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:85:30: error: division by zero [-Werror=div-by-zero]
   85 | #define MAX_TICKS (TIMER_MAX / CYCLES_PER_TICK)
      |                              ^
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:285:15: note: in expansion of macro 'MAX_TICKS'
  285 |  cached_icr = MAX_TICKS * CYCLES_PER_TICK;
      |               ^~~~~~~~~
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:293:8: error: division by zero [-Werror=div-by-zero]
  293 |  ticks /= CYCLES_PER_TICK;
      |        ^~
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c: In function 'sys_clock_driver_init':
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:85:30: error: division by zero [-Werror=div-by-zero]
   85 | #define MAX_TICKS (TIMER_MAX / CYCLES_PER_TICK)
      |                              ^
/home/galak/git/zephyr/drivers/timer/mchp_xec_rtos_timer.c:374:15: note: in expansion of macro 'MAX_TICKS'
  374 |  cached_icr = MAX_TICKS;
      |         
```

galak, 2021-08-18:
@nordic-krch is there a reason to set `CONFIG_SYS_CLOCK_TICKS_PER_SEC=100000` in the prj.conf for this test?

galak, 2021-08-18:
> @nordic-krch is there a reason to set `CONFIG_SYS_CLOCK_TICKS_PER_SEC=100000` in the prj.conf for this test? This breaks the mchp_xec_rtos_timer driver.

This also breaks the ite_it8xxx2_timer driver.

galak, 2021-08-18:
Seeing failures on:

```
ERROR   - intel_adsp_cavs15         tests/lib/ringbuffer/libraries.data_structures     FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/intel_adsp_cavs15/tests/lib/ringbuffer/libraries.data_structures/build.log
INFO    - Total complete:    1/  19   5%  skipped:    0, failed:    1
ERROR   - rcar_h3ulcb_cr7           tests/lib/ringbuffer/libraries.data_structures     FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/rcar_h3ulcb_cr7/tests/lib/ringbuffer/libraries.data_structures/build.log
INFO    - Total complete:    2/  19  10%  skipped:    0, failed:    2
ERROR   - mec172xevb_assy6906       tests/lib/ringbuffer/libraries.data_structures     FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mec172xevb_assy6906/tests/lib/ringbuffer/libraries.data_structures/build.log
INFO    - Total complete:    3/  19  15%  skipped:    0, failed:    3
ERROR   - it8xxx2_evb               tests/lib/ringbuffer/libraries.data_structures     FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/it8xxx2_evb/tests/lib/ringbuffer/libraries.data_structures/build.log
INFO    - Total complete:    4/  19  21%  skipped:    0, failed:    4
ERROR   - mec1501modular_assy6885   tests/lib/ringbuffer/libraries.data_structures     FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mec1501modular_assy6885/tests/lib/ringbuffer/libraries.data_structures/build.log
INFO    - Total complete:    5/  19  26%  skipped:    0, failed:    5
ERROR   - mec15xxevb_assy6853       tests/lib/ringbuffer/libraries.data_structures     FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mec15xxevb_assy6853/tests/lib/ringbuffer/libraries.data_structures/build.log
INFO    - Total complete:    6/  19  31%  skipped:    0, failed:    6
ERROR   - qemu_cortex_m0            tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_m0/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:    7/  19  36%  skipped:    0, failed:    7
ERROR   - qemu_cortex_r5            tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_r5/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:    8/  19  42%  skipped:    0, failed:    8
ERROR   - mps3_an547                tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/mps3_an547/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:    9/  19  47%  skipped:    0, failed:    9
ERROR   - mps2_an521                tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/mps2_an521/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:   10/  19  52%  skipped:    0, failed:   10
ERROR   - mps2_an385                tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/mps2_an385/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:   11/  19  57%  skipped:    0, failed:   11
ERROR   - qemu_arc_em               tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_arc_em/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:   12/  19  63%  skipped:    0, failed:   12
ERROR   - qemu_arc_hs6x             tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_arc_hs6x/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:   13/  19  68%  skipped:    0, failed:   13
ERROR   - qemu_arc_hs               tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_arc_hs/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:   14/  19  73%  skipped:    0, failed:   14
ERROR   - qemu_cortex_a53           tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_a53/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:   15/  19  78%  skipped:    0, failed:   15
ERROR   - qemu_leon3                tests/lib/ringbuffer/libraries.data_structures     FAILED: Timeout
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_leon3/tests/lib/ringbuffer/libraries.data_structures/build.log
INFO    - Total complete:   16/  19  84%  skipped:    0, failed:   16
ERROR   - qemu_nios2                tests/lib/ringbuffer/libraries.data_structures     FAILED: Timeout
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_nios2/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:   17/  19  89%  skipped:    0, failed:   17
ERROR   - qemu_x86_64               tests/lib/ringbuffer/libraries.data_structures     FAILED: Timeout
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_x86_64/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:   18/  19  94%  skipped:    0, failed:   18
ERROR   - nrf52_bsim                tests/lib/ringbuffer/libraries.data_structures     FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/nrf52_bsim/tests/lib/ringbuffer/libraries.data_structures/handler.log
INFO    - Total complete:   19/  19  100%  skipped:    0, failed:   19
INFO    - 0 of 19 test configurations passed (0.00%), 19 failed, 0 skipped with 0 warnings in 69.76 seconds
```

scottwcpg, 2021-08-18:
@galak I tested adding a board level prj config file for this test without the clock config.  The test passed.

KangJianX, 2021-08-19:
It is also build failed on ADSP. So added platform labels.

PerMac, 2021-08-20:
This test builds but fails during execution on nrf52dk_nrf52832 with 
```
START - test_ringbuffer_concurrent
E: ***** USAGE FAULT *****
E:   Attempt to execute undefined instruction
E: r0/a1:  0x200014a0  r1/a2:  0x00000000  r2/a3:  0x20001454
E: r3/a4:  0x000071af r12/ip:  0x00000000 r14/lr:  0x00007f41
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x000055c2
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200004c8 (test_ringbuffer_concurrent)
E: Halting system
```
and on nrf9160dk_nrf9160_ns with a timeout

yerabolu, 2021-08-24:
fails on microchip, upsquared and EHL_CRB as well.

PR #38200 (2021-09-02): Fixes in ring buffer tests
Fixing failing tests for multiple platforms.

Fixes #38093.
Fixes #37768.
closed 2021-09-02 by nashif