zephyrproject-rtos/zephyr #37777: mpu6050 sensor_value_to_double value not printing using printf or uart_poll_out

ric96, 2021-08-18:
Mostly following the `samples/sensor/mpu6050` on a  stm32f411e_disco.

I am unable to print double values either using `printf` directly or using `uart_poll_out` with a char array generated using sprintf.
The printed value always ends up being blank.
For example: `printf("Temp: %f \n", sensor_value_to_double(&temperature);` always results in the following output: `Temp: `

However printing values typecast to int seems fine.
Fpr example: `printf("Temp: %d \n", (int)sensor_value_to_double(&temperature);` results in an output similar to: `Temp: 30`

This is true for all SENSOR_CHAN_ACCEL_XYZ, SENSOR_CHAN_GYRO_XYZ and SENSOR_CHAN_AMBIENT_TEMP.

I have not tested it with another sensor so I am not sure if its an issue with `sensor_value_to_double` or specific to the mpu6050

I can print other double and float values just fine.

avisconti, 2021-09-06:
@ric96                                                                                 
                                                                                
I don't think it is a sensor_value_to_double() issue, because I'm using it extensively for x_nucleo_iks01a3 (and other shields).  It might be an issue related to the mpu6050 sensor, but I do not have it, so I cannot try it out.

avisconti, 2021-09-06:
@ric96                                                                                
Can you investigate more about it? I just saw the driver code and it seems fine to me.

stephanosio, 2021-09-23:
It looks like you are using the newlib (`CONFIG_NEWLIB_LIBC=y`).

Please make sure that `CONFIG_NEWLIB_LIBC_FLOAT_PRINTF` is set to `y`.

stephanosio, 2021-11-30:
Closing since an answer has been provided.

closed 2021-11-30 by stephanosio