zephyrproject-rtos/zephyr #37779: adc sam0 interrupt mapping, RESRDY maps to second interrupt in samd5x.dtsi

photonthunder, 2021-08-18:
When using the ADC, adc_context_wait_for_completion would never complete because the isr (adc_sam0_isr) would never fire.  Turns out that the samd5x.dtsi calls out both interrupt mappings:

`		adc0: adc@43001c00 {
			compatible = "atmel,sam0-adc";
			reg = <0x43001C00 0x4A>;
			interrupts = <118 0>, <119 0>;`

According to the datasheet (Pg 72 of DS60001507G) RESRDY maps to 119 and OVERRUN and WINMON to 118.  In my overlay I just removed 118 and then it worked:

`&adc0 {
	status = "okay";
	compatible = "atmel,sam0-adc";
	label = "ADC_0";
	interrupts = <119 0>;`

I think the actual fix should probably be somewhere else, but I am not familiar enough with the IRQ_CONNECT and other settings to know how to make this a universal fix.  Mostly just wanted to annotate this in case others encounter it.

stephanosio, 2021-08-31:
cc @nandojve @mnkp @benpicco

cfriedt, 2021-08-31:
@photonthunder - could you possibly submit a PR with your proposed fix?

closed 2021-09-02 by cfriedt

photonthunder, 2021-09-02:
@cfriedt, this is one time were being slow to implement a fix is a good thing because I really like the fix by @nandojve...

commit bca28015 (2021-09-02): drivers: adc: adc_sam0: Fix result ready interrupt

Atmel sam0 adc peripheral have multiple interrupt vectors for same5x
devices.  This configure interrupt vector by name to ensure that proper
interrupt handle will be executed.

Fixes #37779