zephyrproject-rtos/zephyr #37781: nucleo_lg96zg lpuart1 driver not working

nwsetec, 2021-08-18:
**Describe the bug**
In Zephyr v2.6.0 when running the hello_world sample printk output is not printed to the console.
printk does print to the console in v2.5.0

I did a git bisect and found commit 888a24f76d2e5fd3bff313f3551af08973940ad6 breaks this functionality.

**To Reproduce**
Steps to reproduce the behaviour:
1. Checkout v2.6.0-branch or main
2. Connect terminal to serial port
3. west build -p -b nucleo_l496zg zephyr/samples/hello_world
4. west flash
5. Observe "Hello World! nucleo_l496zg" is not printed in terminal

**Expected behavior**
"Hello World! nucleo_l496zg" is printed in terminal

**Impact**
annoyance

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain ARM GNU 7.3.1

**Additional context**
@ABOSTM and I discussed this issue briefly on slack

ABOSTM, 2021-08-19:
Issue is due to PWR clock which is disabled when **LL_PWR_EnableVddIO2()** is called (fromgpio_stm32_init()).
@nwsetec can you check #37787 fixes your Jlink use case ? Thanks

nwsetec, 2021-08-19:
I'm not so worried about Jlink (my dev board is now working with openOCD) but more worried about the console/shell working or not.

I tested it with the hello world and shell module samples and it does seem to work.  If possible we were hoping for a v2.6.0-branch backport?

ABOSTM, 2021-08-19:
I'm not familiar with zephyr backport.
@galak @cfriedt , can you help me to know what it means, and what should be done ?

PR #37787 (2021-08-19): drivers: clock_control: stm32: enable PWR clock unconditionally
Enable PWR clock unconditionally for L4, L5 and U5
like it is done on other stm32 series

This is a regression introduced by #32390

Fixes #37781
commit 482198bf (2021-08-19): drivers: clock_control: stm32: enable PWR clock unconditionally

Enable PWR clock unconditionally for L4, L5 and U5
like it is done on other stm32 series

Fixes #37781
PR #38163 (2021-09-18): drivers: clock_control: stm32: enable PWR clock unconditionally
Enable PWR clock unconditionally for L4, L5 and WB/L
like it is done on other stm32 series

Fixes #37781
commit 25771e69 (2021-09-18): drivers: clock_control: stm32: enable PWR clock unconditionally

Enable PWR clock unconditionally for L4, L5 and U5
like it is done on other stm32 series

Fixes #37781
closed 2021-08-19 by galak