zephyrproject-rtos/zephyr #37786: Example for tca9546a multiplexor driver

gabbas1, 2021-08-19:
Hello guys @kurddt @carlescufi ,

I was wondering if there is a minimal working **example for the new tca9546a** driver, because at the moment I dont know how to make it work.

I am using the **nrf52840-DK with Zephyr (latest)** to talk to SHT31. Now I want to use the TCA inbetween my DK and multiple SHT31. But I already struggle to switch channels.

Usually I add my SHT to the devicetree with an overlay like this:

```
 &i2c0 {
	sht3xd@44 {
		compatible = "sensirion,sht3xd";
		reg = <0x44>;
		label = "SHT3XD";
		#alert-gpios = <&gpio1 10 GPIO_ACTIVE_HIGH>;
	};
};
```

But how does it work with the tca9546a?
Do I have to add my sensors as channels of the TCA, like it can be found in `zephyr/tests/drivers/i2c/i2c_tca9546a?

```
&i2c0 {
	status = "okay";
	mux: tca9546a@70 {
		compatible = "ti,tca9546a";
		reg = <0x70>;
		status = "okay";
		label = "i2c_mux";
		#address-cells = <1>;
		#size-cells = <0>;

		ch0: mux_i2c@0 {
			label = "mux_dw_0";
			reg = <0>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";
		};

		ch1: mux_i2c@1 {
			label = "mux_dw_1";
			reg = <1>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";
        };
	};
};
```

And do I have the get device binding in my main.c 
` const struct device *i2c_muxer = device_get_binding("i2c_mux"); ` just for the TCA or for the TCA and the attached sensors behind the TCA and how do I switch channels?!
I am a bit confused about how this works.

Thanks for your driver and help :+1: 
Sebastian

kurddt, 2021-08-19:
Hi @gabbas1 The only thing you need to do is to add your sensor as a child of the channel. For instance if you have two SHT31 sensors plugged on channel 0 and 1 the device tree would look like:
```
&i2c0 {
	status = "okay";
	mux: tca9546a@70 {
		compatible = "ti,tca9546a";
		reg = <0x70>;
		status = "okay";
		label = "i2c_mux";
		#address-cells = <1>;
		#size-cells = <0>;

		ch0: mux_i2c@0 {
			label = "mux_dw_0";
			reg = <0>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";

			sht3xd@44 {
				compatible = "sensirion,sht3xd";
				reg = <0x44>;
				label = "SHT3XD_1";
				#alert-gpios = <&gpio1 10 GPIO_ACTIVE_HIGH>;
			};
		};

		ch1: mux_i2c@1 {
			label = "mux_dw_1";
			reg = <1>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";
			sht3xd@44 {
				compatible = "sensirion,sht3xd";
				reg = <0x44>;
				label = "SHT3XD_2";
				#alert-gpios = <&gpio1 10 GPIO_ACTIVE_HIGH>;
			};
        	};
	};
};
```

kurddt, 2021-08-19:
This is described in https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/i2c/ti%2Ctca9546a.yaml

gabbas1, 2021-08-19:
> This is described in https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/i2c/ti%2Ctca9546a.yaml

Thank you. Sorry, haven't seen that description.

And in the main.c I can just use my code for SHT31 like the TCA wouldn'n be there?

``` c
...
const struct device *dev = device_get_binding("SHT3XD");
	int rc;
	uint32_t cnt = 0;

	if (dev == NULL) {
		printf("Could not get SHT3XD device\n");
		return;
	}

	while (true) {
		struct sensor_value temp, hum;

		rc = sensor_sample_fetch(dev);
...
```

kurddt, 2021-08-19:
Yes that's right

gabbas1, 2021-08-19:
Okay i tried it with my (working) the SHT31 example, with is basically the sample which comes with Zephyr.

I just edited the devicetree to look like this: (just one sht for now)

``` 
&i2c0 {
	status = "okay";
	mux: tca9546a@70 {
		compatible = "ti,tca9546a";
		reg = <0x70>;
		status = "okay";
		label = "i2c_mux";
		#address-cells = <1>;
		#size-cells = <0>;

        ch0: mux_i2c@0 {
			label = "mux_dw_0";
			reg = <0>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";

            sht1: sht3xd@44 {
                status = "okay";
        		compatible = "sensirion,sht3xd";
        		reg = <0x44>;
        		label = "SHT3XD";
        		#alert-gpios = <&gpio1 10 GPIO_ACTIVE_HIGH>;
        	};
		};

/*
		ch1: mux_i2c@1 {
			label = "mux_dw_1";
			reg = <1>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";
        };
*/
	};
};

/{
	aliases {
        i2c-channel-0 = &sht1;
	};
};
```

To the pjr.conf-file I added the `CONFIG_I2C_TCA9546A=y`:

```
#CONFIG_LOG=y

CONFIG_I2C=y
CONFIG_SENSOR=y
CONFIG_SHT3XD=y
CONFIG_SENSOR_LOG_LEVEL_DBG=y

CONFIG_SERIAL=y
CONFIG_UART_CONSOLE=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_I2C_TCA9546A=y

CONFIG_PM=y
```

The beginning of my main.c looks like this:

``` c
#include <zephyr.h>
#include <device.h>
#include <drivers/sensor.h>
#include <stdio.h>

void main(void)
{
	const struct device *dev = DEVICE_DT_GET(DT_ALIAS(i2c_channel_0));
	int rc;

	if (dev == NULL) {
		printf("Could not get SHT3XD device\n");
		return;
	}

	while (true) {
		struct sensor_value temp, hum;

		printf("1\n");
		rc = sensor_sample_fetch(dev);
		printf("2\n");
...
```

The code stops after `printf("1\n");`. 
The error occures in the driver `tca9546a.c` in the function `tca9546a_channel_init`, because via uart i receive the message:
`I2C mux root i2c_mux not ready`
after the:

``` c
if (!device_is_ready(cfg->root)) {
		printf("I2C mux root %s not ready\n", cfg->root->name);
		return -ENODEV;
	}
```  

Any idea what could be the problem?

PS: my wiring should be correct, because I can get samples from the same circuit (the sht31 with tca9546 in between) with an arduino

kurddt, 2021-08-19:
Can you make sure that CONFIG_I2C_TCA9546_CHANNEL_INIT_PRIO > CONFIG_I2C_TCA9546_ROOT_INIT_PRIO > {init priority used by the i2c driver priority}

gabbas1, 2021-08-19:
Well, at the moment they seam to use both the default value.

```
config I2C_TCA9546_ROOT_INIT_PRIO
	int "TCA9546a root driver init priority"
	default I2C_INIT_PRIORITY

config I2C_TCA9546_CHANNEL_INIT_PRIO
	int "TCA9546a channel driver init priority"
	default I2C_INIT_PRIORITY
```

And I noticed (with printf-debugging) that `tca9546a_channel_init` is executed before `tca9546_root_init`. That is probably not good, right?!

kurddt, 2021-08-19:
Yes they need to be modified. Try with CONFIG_I2C_TCA9546_ROOT_INIT_PRIO=61 and CONFIG_I2C_TCA9546_CHANNEL_INIT_PRIO=62

gabbas1, 2021-08-19:
It works! (at least with that one sensor for now)
Nice!
Dont know if I ever found that out on my own, so thanks a lot!

closed 2021-08-19 by gabbas1

gabbas1, 2021-08-20:
Hey,
one more question @kurddt,

everything works fine, as long as I use just one sensor of a kind, i.e. one sht31 and one sht21. So like this:

```
&i2c0 {
	status = "okay";
	mux: tca9546a@70 {
		compatible = "ti,tca9546a";
		reg = <0x70>;
		status = "okay";
		label = "i2c_mux";
		#address-cells = <1>;
		#size-cells = <0>;

        ch0: mux_i2c@0 {
			label = "mux_dw_0";
			reg = <0>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";

            sht1: sht3xd@44 {
                status = "okay";
        		compatible = "sensirion,sht3xd";
        		reg = <0x44>;
        		label = "SHT3XD";
        	};
		};

		ch1: mux_i2c@1 {
			label = "mux_dw_1";
			reg = <1>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";

            sht2: sht21@40 {
                status = "okay";
        		compatible = "sensirion,sht21";
        		reg = <0x40>;
        		label = "SHT21";
        	};
        };
	};
};
```

But when I wanna use i.e. two sht21 (or two sht31, doesn't matter) like this:

```
&i2c0 {
	status = "okay";
	mux: tca9546a@70 {
		compatible = "ti,tca9546a";
		reg = <0x70>;
		status = "okay";
		label = "i2c_mux";
		#address-cells = <1>;
		#size-cells = <0>;

        ch0: mux_i2c@0 {
			label = "mux_dw_0";
			reg = <0>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";

            sht1: sht21_1@40 {
                status = "okay";
        		compatible = "sensirion,sht21";
        		reg = <0x40>;
        		label = "SHT21_1";
        	};
		};

		ch1: mux_i2c@1 {
			label = "mux_dw_1";
			reg = <1>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";

            sht2: sht21_2@41 {
                status = "okay";
        		compatible = "sensirion,sht21";
        		reg = <0x40>;
        		label = "SHT21_2";
        	};
        };
	};
};
```
I get this errors in the linking process:

```
git/zephyrproject1/zephyr/samples/imms/sht21_tca9546a/build/../src/main.c:36: undefined reference to `__device_dts_ord_70'
```

Can I not have multiple instances of a sensor/driver?

With best regards,
Sebastian

```
...
[152/159] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map 
: && ccache Programme/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr/zephyr_prebuilt.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=git/zephyrproject1/zephyr/samples/imms/sht21_tca9546a/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/sensor/sht21/libdrivers__sensor__sht21.a  zephyr/drivers/serial/libdrivers__serial.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/segger/libmodules__segger.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"Programme/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp"  -L/home/smiethe/git/zephyrproject1/zephyr/samples/imms/sht21_tca9546a/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd git/zephyrproject1/zephyr/samples/imms/sht21_tca9546a/build/zephyr && /usr/bin/cmake -E echo
/Programme/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `main':
git/zephyrproject1/zephyr/samples/imms/sht21_tca9546a/build/../src/main.c:36: undefined reference to `__device_dts_ord_70'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build git/zephyrproject1/zephyr/samples/imms/sht21_tca9546a/build
```

kurddt, 2021-08-20:
I can't find any driver compatible with `sensirion,sht21` in zephyr repo, so I can't really answer you.

Also sht2 `reg` (`reg = <0x40>;`) and  unit address (`sht21_2@41`) does not match

kurddt, 2021-08-20:
If you are using code from this old PR: https://github.com/zephyrproject-rtos/zephyr/pull/22862, Then no you can't have two device using the driver

gabbas1, 2021-08-20:
Yes, i use parts of the code from this old PR.
But Why is it not possible?

Nevertheless, when using the SHT3xD, the same error occurs:

```
&i2c0 {
	status = "okay";
	mux: tca9546a@70 {
		compatible = "ti,tca9546a";
		reg = <0x70>;
		status = "okay";
		label = "i2c_mux";
		#address-cells = <1>;
		#size-cells = <0>;

        ch0: mux_i2c@0 {
			label = "mux_dw_0";
			reg = <0>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";

            sht1: sht3xd@44 {
                status = "okay";
        		compatible = "sensirion,sht3xd";
        		reg = <0x44>;
        		label = "SHT3XD";
        	};
		};

		ch1: mux_i2c@1 {
			label = "mux_dw_1";
			reg = <1>;
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;
			compatible = "ti,tca9546a-channel";

            sht2: sht3xd@44 {
                status = "okay";
        		compatible = "sensirion,sht3xd";
        		reg = <0x44>;
        		label = "SHT3XD";
        	};
        };
	};
};
```

kurddt, 2021-08-20:
The SHT3xD driver have the same issue it can only handles one device. 

To handle several devices the driver need to be modified. It might be just a matter of changing `DEVICE_DT_INST_DEFINE` to `DT_INST_FOREACH_STATUS_OKAY`

gabbas1, 2021-08-20:
Okay, I needed to change in the driver `sht21.c`:

``` c
static struct sht21_data sht210_driver;

static const struct sht21_config sht210_cfg = {
	.bus_name = DT_INST_BUS_LABEL(0),
	.base_address = DT_INST_REG_ADDR(0)
};

DEVICE_DT_INST_DEFINE(0, sht21_init, NULL,
		    &sht210_driver, &sht210_cfg,
		    POST_KERNEL, CONFIG_SENSOR_INIT_PRIORITY,
		    &sht21_driver_api);
```

to:

``` c
#define SHT21_INIT(index)										\
	static struct sht21_data sht210_driver_##index;				\
																\
	static const struct sht21_config sht210_cfg_##index = { 	\
		.bus_name = DT_INST_BUS_LABEL(index),					\
		.base_address = DT_INST_REG_ADDR(index)					\
	};															\
																\
	DEVICE_DT_INST_DEFINE(index, sht21_init, NULL,				\
			    &sht210_driver_##index, &sht210_cfg_##index,	\
			    POST_KERNEL, CONFIG_SENSOR_INIT_PRIORITY,		\
			    &sht21_driver_api);

DT_INST_FOREACH_STATUS_OKAY(SHT21_INIT)
```

and well it works now with multiple instances of the driver!
(for SHT3xD it is the same)

Thanks!

kurddt, 2021-08-23:
Would be good if you can submit a PR for those changes